thp/psmoveapi #332: raspberry pi 3 CMake Error

AvadaX, 2017-11-22:
Hi，I tried to build psmove api on Raspberry pi 3,and I followed the [Building for the Pocket C.H.I.P](http://psmoveapi.readthedocs.io/en/latest/build.html#building-for-the-pocket-c-h-i-p)

but when I try bash -e -x scripts/pocketchip/build.sh, it shows:

```
CMake Error at src/CMakeLists.txt:163 (add_library):
  Cannot find source file:

    /home/pi/psmoveapi/external/hidapi/linux/hid.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
Call Stack (most recent call first):
  CMakeLists.txt:6 (include)

CMake Error: CMake can not determine linker language for target: psmoveapi
CMake Error: Cannot determine link language for target "psmoveapi".
-- Generating done
-- Build files have been written to: /home/pi/psmoveapi/build
```
[image]

nitsch, 2017-11-22:
Are there any files in the /home/pi/psmoveapi/external/hidapi directory?
If there are not, you probably did not clone the repository using the
--recursive option, so the submodules are missing.

To fix this, cd into /home/pi/psmoveapi and run the command

  git submodule update --init

This should fetch the required submodules.

AvadaX, 2017-11-23:
thank you, @nitsch ! , it works

closed 2017-11-23 by AvadaX

zero-point, 2018-03-18:
Sorry to reignite this, but I had the same error and it took me a long time to find this fix. Shouldn't this instruction "git submodule update --init" be put in the documentation?