thp/psmoveapi #335: unable to build on raspberry pi with release 4.0.4

T-Lancer, 2017-11-30:
I am attempting to build from source on Raspian Strech using release 4.0.4.

the build fails during manual build using the supplied instructions.  I have the following output of cmake and make:

[errorlog_pi_psmoveapi.txt](https://github.com/thp/psmoveapi/files/1518368/errorlog_pi_psmoveapi.txt)

nitsch, 2017-12-02:
This seems to be an issue with OpenCV. Which version of OpenCV are you using?

(You can also clone and build psmoveapi's current master branch from the Git repository. It should print that additional library information during the initial detection step.)

T-Lancer, 2017-12-04:
thanks for the reply. I'm attempting to compile with OpenCV 3.3.1 also compiled from source using the following tutorial but using 3.3.1 rather than 3.3.0 and without using any virtual enviroments: https://www.pyimagesearch.com/2017/09/04/raspbian-stretch-install-opencv-3-python-on-your-raspberry-pi/

I again attempted to compile after cloning from the master branch and have the following output:
[errorlog_pi_psmoveapi_master.txt](https://github.com/thp/psmoveapi/files/1526294/errorlog_pi_psmoveapi_master.txt)

it seems to still have problems with cvcapture.

I have also attempted to compile on an Ubuntu x64 based system which has OpenCV 3.3.0 installed (also from source) and received the same error when compiling.

thanks again for the prompt reply.

nitsch, 2017-12-04:
The problem is that OpenCV 3 deprecated the C API which psmoveapi is using. They are not maintaining it any longer, so stuff most likely got broken with new releases in the 3.x branch.

Since there isn't anything psmoveapi needs from OpenCV 3, you should use a version from OpenCV's 2.x branch. The most recent 2.4.13.4 seems to work just fine. But older versions are also be suitable.

nitsch, 2017-12-04:
@T-Lancer BTW, if OpenCV 2.x is too much hassle for you to set up and you have OpenCV 3.1.0 available, that version should still work.

T-Lancer, 2017-12-06:
thanks.  compiling with OpenCV2 worked.  not sure if this is still the right place to ask or if I'll open a new ticket, but there seems to be a real lack of documentation in regards to getting any of the examples working.  for example, where is the psmove module for python located within the srouce files?  I am unable to run any of the python examples as the scripts fail when trying to import the module psmove.

nitsch, 2017-12-06:
> there seems to be a real lack of documentation in regards to
> getting any of the examples working.

True, this topic seems to have been largely neglected. If you finally get
things working with Python, some help with the documentation would be
greatly appreciated. Maybe you can extend the documents found in docs/
with some information regarding Python and/or the examples.

> for example, where is the psmove module for python located
> within the srouce files?

Python bindings (among others) are built from bindings/psmove.i using SWIG.

There have been quite a few questions regarding Python in the past. Please
search the list of issues here on GitHub (do not forget the closed ones).
If that does not help, I suggest opening a new issue with your specific
problem, error messages etc.

thp, 2017-12-09:
The Python examples are laid out so that when you build the CMake solution in the customary ˋbuildˋ folder in the PSMove API checkout, and have Swig and Python Dev installed, the Python examples should find the modules without needing to install anything (ˋmake installˋ). Let me know if that stopped to work.

T-Lancer, 2017-12-11:
thanks for the reply.

currently this are my steps:

complie openCV -> compile psmoveapi -> place psmove.i in a folder with files from /include and /build (psmove_config.h) -> "swig -python psmove.i "-> output is psmove.py -> from there I try to run python pair.py and receive the following import error: (note: psmove.py and pair.py are placed in the /build folder and the paths to all files are adjusted as to point to psmove.py)

ImportError: no modul names _psmove

if I do not build the module with swig I don't even get that far.  what might I be doing wrong?

thp, 2017-12-11:
Use `cmake` to build it, there are instructions here:

http://psmoveapi.readthedocs.io/en/latest/build.html#building-on-ubuntu-14-04

thp, 2017-12-11:
If you really want to do it manually, you need to compile the `.c` source file that `swig` generates, this will generate the `_psmove.so` dynamic Python module that `psmove.py` relies on.

This is what CMake does under the hood (shortened/simplified):

    swig -python -outdir build -o psmovePYTHON_wrap.c psmove.i
    cc -shared -lpsmoveapi -o _psmove.so psmovePYTHON_wrap.c

But again, the recommended way is `cmake`.

Note that there are also non-swig bindings now in https://github.com/thp/psmoveapi/blob/master/bindings/python/psmoveapi.py that just need the shared library and uses `ctypes` to access it. Those new bindings have a different API than the old one, though.

T-Lancer, 2017-12-12:
thanks,  I have been attempting to use cmake to build from source.  however something must still be going wrong.

having installed swig I decided to try the automatic build script for Debian (as raspbian is also debian-based).  here I do get further then previous attempts but now I have problems with the libpsmoveapi_tracker.so library.

Output is attached:
[errorlog_with_swig.txt](https://github.com/thp/psmoveapi/files/1550695/errorlog_with_swig.txt)

thp, 2017-12-12:
Did you do an `apt install libv4l-dev` (or `apt-get install libv4l-dev`, depending on how recent your Raspbian is)?

T-Lancer, 2017-12-12:
yes, version 1.12.3-1 is installed

T-Lancer, 2017-12-12:
I have sucessfully built psmoveapi using the debian install script.  I had to add a line to the src/tracker/CMakeLists.txt file on line 86: 
`list(APPEND PSMOVEAPI_TRACKER_REQUIRED_LIBS v4l2)`

issue77 referes to the same problem: https://github.com/thp/psmoveapi/issues/77

after this, I attempted to run the pair.py example again.  again an importError stopped my progress:

> pi@raspberrypi:~/Downloads/psmoveapi-4.0.4/examples/python $ python pair.py
> Traceback (most recent call last):
>   File "pair.py", line 35, in <module>
>     import psmove
>   File "../../build/psmove.py", line 21, in <module>
>     _psmove = swig_import_helper()
>   File "../../build/psmove.py", line 20, in swig_import_helper
>     return importlib.import_module('_psmove')
>   File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
>     __import__(name)
> ImportError: dynamic module does not define init function (init_psmove)

nitsch, 2017-12-13:
Please use the current master branch from the Git repository. It already includes the v4l fix which was only added shortly after release 4.0.4.

Regarding the Python issue: Please see #328. Again, build the current master which should show you which version of Python you are building against (in the CMake configure step) and make sure that is the version you are running the scripts with afterwards.

T-Lancer, 2017-12-13:
Indeed.  certainly getting there.  It seems to build bindings for python3.  though I intend to use python2.7.  However, trying to run the script with python 3 gives me errors as well.  I will attempt to build for python2.7.

[master_log.txt](https://github.com/thp/psmoveapi/files/1555263/master_log.txt)

nitsch, 2017-12-13:
I think your Python 3 run actually works already, but the script does not handle errors very well.

Try connecting your Move controller via USB and enable Bluetooth on your Raspberry Pi.

nitsch, 2017-12-15:
@T-Lancer I have added some documentation and a couple of improvements that should help fix issues like the ones you were having more easily. See #337. Do you have any additions that might further improve things?

T-Lancer, 2017-12-18:
I'll have to keep testing.  But so far I have compiled with python 2.7 and python 3 bindings. So that seems to work well enough now.  Some python examples seem to work. Such as pairing and LED tests. In regards to acutal bluetooth pairing, I'm still haveing some problems there. "pairing" returns sucess. However, when disconnection from USB, the controller appears to sometimes be connected, yet no example or direction psmove command actually work.  running "psmove list" lists only the bluetooth device.

When trying to run "psmove calibrate".  I receive the error of having a bad calibration file and that non exits.  running "psmove pair" tells me I have not calibrated the controller yet.  So I have this loop of being unable to calibrate.  this is reflected in some of the python examples not working correctly.

[psmove_pair.txt](https://github.com/thp/psmoveapi/files/1567932/psmove_pair.txt)

For example, running test_calibration returns the following error:

[test_calibrationt.txt](https://github.com/thp/psmoveapi/files/1567933/test_calibrationt.txt)

further more, attempting to run a tracker example returns this error:

[test_tracker.txt](https://github.com/thp/psmoveapi/files/1567935/test_tracker.txt)

if it makes sense, I'd open a new issue regarding pairing issues and the python examples.

nitsch, 2017-12-19:
> if it makes sense, I'd open a new issue regarding pairing issues and the python examples.

Please do. Keeping issues separated (pairing vs. calibration vs. Python)
also makes them easier to handle and search/find later.

closed 2017-12-20 by T-Lancer