thp/psmoveapi #336: Explicitly use OpenCV 2

nitsch, 2017-12-04:
We should use a version from OpenCV's 2.x branch. The C API was deprecated in OpenCV 3.x, and stuff breaks (see #335, #251). Plus, there really is no need to use OpenCV 3. We use non of its features. The 2.x branch is still maintained by OpenCV folks.

If there are mentions of OpenCV 3 in the documentation, they should be replaced. If some of the build scripts either use OpenCV 3 or do not specifically use 2.x, they should be fixed, too.

thp, 2017-12-09:
Yes, is there any good branch in upstream OpenCV that we could reference?

nitsch, 2017-12-09:
> Yes, is there any good branch in upstream OpenCV that we could reference?

Sure, branch 2.4 (https://github.com/opencv/opencv/tree/2.4) would be
the one.

commit 23a57059 (2017-12-30): Merge pull request #339 from thp/use-opencv-2

Use OpenCV 2.4 (Fixes #336)
commit b4560905 (2017-12-30): Use OpenCV 2.4 (Fixes #336)
closed 2017-12-30 by thp