thp/psmoveapi #340: Missing libpsmove_java.dll?

GeorgeTsakiridis, 2017-12-16:
Hi, I downloaded the source code and executed the build_msvc.bat with the arguments "2017 x64". However build fails with this:
`Error copying file "C:/Users/GeorgeProgramming/Desktop/psmoveapi/build-x64/libpsmove_java.dll" to "C:/Users/GeorgeProgramming/Desktop/psmoveapi/bindings/processing/libraries/psmove/library/windows/libpsmove_java.dll"` . 
Searching the folder this file is indeed missing. Am I doing anything wrong? Here is a detailed step by step of what I did:
1)Downloaded psmoveapi source from GitHub
2)Downloaded hidapi, libusb, PS3EYEDriver source and extracted it to the external folder.
3)Opened "Development Command Prompt for VS 2017"
4)Executed "build_msvc.bat 2017 x64"
5)Got the above error

Here is the whole log: https://pastebin.com/PiTGhmMe
Thank you for your time

nitsch, 2017-12-16:
The DLL was built in `build-x64\Debug\` but the generated Visual Studio solution somehow expects it in `build-x64\`. 

I won't be able to test this on a Windows machine before Monday. Maybe someone else can look into this earlier?

GeorgeTsakiridis, 2017-12-16:
Well I don't know if it is normal or not but the `libpsmove_java.dll` file does not exist anywhere. I couldn't find it at `build-x64\Debug\` and doing a search returned nothing. Maybe it wasn't generated at all?

nitsch, 2017-12-16:
That output directory was merely what I understood your build log was saying about the actual location of that DLL.

You could build it on Linux, but you could then not use it on Windows, if that was your question. The libraries that the Java binding is using are platform-specific.

GeorgeTsakiridis, 2017-12-17:
OK so I tried building it in Ubuntu and it worked. I got a .jar file. But now there whenever I try to run any PSMove controller code it gives an Assertion Error. For example `        System.out.println(psMove.get_serial());` 
prints null and returns a `[PSMOVE CRITICAL] Assertion fail in psmove_get_serial: move != NULL` error.
The controller is correctly paired with the system as all the examples run perfectly.

nitsch, 2017-12-18:
@GeorgeTsakiridis The Windows build of the Java bindings does indeed have some CMake-related problems that need fixing. But I was not able to reproduce your particular issue. The DLL is indeed built in `build-x64\` just as your log file claimed, too.

Which version of CMake are you using? Are you running the script from the Windows CMD shell or from something else (such as the Git Bash)? Which version of Windows are you using?

GeorgeTsakiridis, 2017-12-18:
I have CMake version 3.10.1, I am running the script from 'Developer Command Prompt for VS 2017' because it has access to 'cl' (I think that's a c++ compiler command) that normal CMD doesn't and I am using Windows 10 64bit (Build 16299)

nitsch, 2017-12-19:
Can you give the following fix a try on your Windows machine?

Clone the repository

    git clone --recursive --branch fix-java
https://github.com/nitsch/psmoveapi/
cd psmoveapi

and then run the build_msvc.bat script as you did before.

GeorgeTsakiridis, 2017-12-19:
Unfortunately I still get the same error...

Here's the log: https://pastebin.com/k3CKtG5R

nitsch, 2017-12-20:
Ugh, I was just being stupid. The problem is related to the **Processing** bindings. The output paths are hard-coded in the CMake file and do not take into account the additional "Debug" and "Release" directories that Visual Studio adds by default.

If you do not need the Processing bindings, you could quickly fix this issue by changing the default value of option *PSMOVE_BUILD_PROCESSING_BINDINGS* in `bindings/CMakeLists.txt` to *OFF*.

I will fix this as soon as I have access to a Windows machine again (might take a couple of days).

nitsch, 2017-12-27:
@GeorgeTsakiridis I updated the fix-java branch. This should fix the problems with the Processing build. Please give it a try.

GeorgeTsakiridis, 2017-12-27:
Great! The build now completes without an error and a .jar file is created. However when adding it to Project it seems like some classes are missing. To be more precise 'PSMoveTracker' and 'Status' showed a missing class error. Also searching for 'libpsmove_java.dll' returned no results...

Here's the log: [https://pastebin.com/JMidCcPG](https://pastebin.com/JMidCcPG )

nitsch, 2017-12-27:
You mentioned earlier that you had it working on Ubuntu. Are the missing
classes a problem only on Windows? If so, you probably only need to fix
your project's search path.

GeorgeTsakiridis, 2018-03-19:
Building with the latest version worked as expected.

closed 2018-03-19 by GeorgeTsakiridis