thp/psmoveapi #341: Language bindings missing from Windows CI build

nitsch, 2017-12-16:
Bindings for Python etc. are built on Linux and Mac OS but not on Windows. It would be nice to have them on Windows, too. It would help catching issues such as #340.

rafaellago, 2019-05-14:
Tried building with mingw64 on linux. And it hasn't generated the .jar or the necessary dll to load psmove in java. Is there some flag that I need to enable to generate those files?

nitsch, 2019-05-14:
> Tried building with mingw64 on linux. And it hasn't generated the .jar or the necessary dll to load psmove in java. Is there some flag that I need to enable to generate those files?

There is a flag, but it is already enabled by default. Maybe some of the 
dependencies are not met.

But please do not hijack this issue here for your question. Please open 
a new issue for this and also include the output of your CMake configure 
step for further inspection.

rafaellago, 2019-05-14:
Sorry. I posted here because I thought it was related. Will open a new one

nitsch, 2020-09-05:
Fixed in 5733cee5f45f268c10bd652233fc1a55b2a115cb.

closed 2020-09-05 by nitsch