thp/psmoveapi #342: unable to properly pair Controllers using psmove binary

T-Lancer, 2017-12-20:
After solving my initial build and python binding problems in issue #335 , I am now trying to get the actual controllers to work properly.  I have compiled psmove on a RaspberryPi 3 on Raspian using Python2.7 bindings.

Currently, when I have a move controller connected via USB I can run "sudo psmove pair" and receiver the following output:

[psmove_pair.txt](https://github.com/thp/psmoveapi/files/1574930/psmove_pair.txt)

It is not clear if a connection is now established.  However when disconnecting the controller from USB it seems there is no connection as the controller instantly switches off.  When powering the controller up again it attemps to connect but times out and turns off.  Clearly something is not working.  While still connected to USB, I can run "sudo psmove calibrate" and receiver the following output:

[psmove_calibrate.txt](https://github.com/thp/psmoveapi/files/1574963/psmove_calibrate.txt)

as the error suggests, I have no such files in either of the location.  However paring doesn't seem to create any files either so that I am stuck in a loop of being unable to do anything.

Running "sudo psmove list" returns the following: (while connected via USB)

[psmove_list.txt](https://github.com/thp/psmoveapi/files/1574990/psmove_list.txt)

again, these files do not exist but there seems to be no way to create them via the psmove calibrate function.

Running "sudo psmove responsivness" makes the controller vibrate every few seconds (while connected via USB). So it can communicate with the controller but no button presses change the output.
[psmove_responsiveness.txt](https://github.com/thp/psmoveapi/files/1575123/psmove_responsiveness.txt)

Moving on the the pairig example for python, I figured maybe this example would yield other results.

running "sudo python2 pair.py" returns the following:

[pair.py.txt](https://github.com/thp/psmoveapi/files/1575009/pair.py.txt)

this shows promise, However, disconnecting the controller results in the same behavior as described above. Running LED tests, such as cops.py or reg_green.py also make the LEDs change and the vibration motor operate.  however this also seemed to be the case without running the python pairing script.  Again, disconnecting the controller resulted in a total loss of communication.

when trying to run test_calibration.py I receiver the following output:

[test_calibrationt.txt](https://github.com/thp/psmoveapi/files/1575133/test_calibrationt.txt)

This may be due to the "missing" calibration files. but again I can't be sure of it.

I could imagine, if pairing is sucessful, most other problems could resolve themselves as well.

nitsch, 2017-12-26:
I suggest you start again fresh and take one step at a time.

Remove the files from `/etc/psmoveapi/` and the files in `/var/lib/bluetooth` that are related to your Move controller (as identified by their BDADDR below the directory that belongs to your Bluetooth adapter). Do not connect any controllers yet.

Now make sure the Bluetooth adapter is enabled on your host machine by running `psmove list`. This should list the adapter but no controllers yet.

Connect a Move controller via USB. To set up a Bluetooth connection, run `sudo psmove pair`. This writes the host BDADDR into the Move controller and sets up some necessary config files on the host: a calibration file for your controller in `/etc/psmoveapi` and a new directory below your adapter's directory in `/var/lib/bluetooth`. It should also restart your Bluetooth service to apply the changes. Check the console output to see whether this step succeeded.

Now disconnect the USB cable from the Move controller and press the PS button on the controller. Its red status LED should start blinking and eventually remain lit, indicating an active Bluetooth connection with the host. Run `psmove list` again to verify that.

With a Bluetooth connection established, you can now run `psmove calibrate`, `example` or any other of the provided applications, including the Python examples.

Next time you want to establish the Bluetooth connection with this controller, you only need to press its PS button again. Only rerun the pairing step if you want to pair the controller to a different Bluetooth adapter.

If, for some reason, you want to perform the pairing using Python, run `examples/python/pair.py` instead of `psmove pair`. It does the same thing.

thp, 2017-12-29:
Also please check if Bluetooth is turned on and running, e.g. `hcitool dev`.

nitsch, 2018-01-11:
@T-Lancer Did this solve your problems?

T-Lancer, 2018-01-15:
@nitsch @thp Thank you for replying.  I have been busy over the christmas break and involed with other projects.  I will however be working in this again this week so I hope I'll have results that I can post.  potentially also to update the current documentation of the API.

nitsch, 2018-11-27:
No response for almost a year. Assuming it is either solved or irrelevant now.

closed 2018-11-27 by nitsch