thp/psmoveapi #346: raspberry pi error:dynamic module does not define init function

AvadaX, 2018-01-25:
Hi, I bulit the psmove api on my raspberry pi 3 today followed the tutorial [building-for-the-pocket-c-h-i-p](https://psmoveapi.readthedocs.io/en/latest/build.html#building-for-the-pocket-c-h-i-p)

after built the psmoveapi, I run the command:
`/home/pi/psmoveapi/build/example`
the PSmove worked

but when I try to run the python example 
`python /home/pi/psmoveapi/examples/python/psmove_test.py`
it showed an error:

[image]

the raspbian system is the latest version
￼￼￼
[image]
and my python version is 2.7.3

how can I solve this?

nitsch, 2018-01-25:
Did you follow the [instructions for building Python bindings](https://psmoveapi.readthedocs.io/en/latest/build.html#python-bindings)?

Which version of psmoveapi are you building? Please use the latest release.

nitsch, 2018-04-20:
@AvadaX Any updates on this?

nitsch, 2018-12-25:
No response for more than 6 months. Assuming it is either solved or irrelevant now.

closed 2018-12-25 by nitsch