thp/psmoveapi #350: Build error, problem on script

Zangetsu38, 2018-03-17:
I have try build with script on VS 2017 X64 but not works in alreacdy, patch static for libusb missing part for 2017, and after not found msbuild not idea why. 
[image]

Zangetsu38, 2018-03-17:
ok i have founded why, need use Visual studio command devlopper, but big probleme it is, build with this, modify few file and after git is dirty, maybe need change change, with simple create sln etc, and test on cmake not works, not found opencv not idea why 
[image]

nitsch, 2018-03-17:
I have trouble understanding you. Please upload a log file containing
the console output produced by build_msvc.bat.

Are you building the current master branch from GitHub or any of the
releases (if so, which one)?

Zangetsu38, 2018-03-17:
It is last master, and i have say after is good compile works with using Visual Studio command, but compile have problem with dirty git, like you see, need improve this system for can compile him without broken/edit file inside git

closed 2018-12-25 by nitsch