thp/psmoveapi #353: Unable to see/pair newest PSMove controllers (w/micro USB)

ChainmailBuddha, 2018-05-07:
Hello,
I currently have the psmoveapi on a Raspberry Pi 3.
This was built/installed as part of the install of JoustMania (https://github.com/adangert/JoustMania), though I don't believe that is necessarily relevant to the issue I am seeing.

I have several of the newest PSMove controllers (with Micro USB connections), and they do not appear to be detected by or able to be paired by the psmoveapi. I have run **psmove list** and confirmed that my bluetooth adapter (TBW-106UB)  is being detected correctly.  However when I run **psmove pair** it returns the message **"Connected controllers: 0"**

I have confirmed that the controllers are fully charged, and when connected to a Windows 10 computer with the same cable, the controllers are connected and installed as expected.

Are these newest controllers supported by psmoveapi, or am I doing something incorrectly?

ChainmailBuddha, 2018-05-07:
Doing some more research on this issue, this seems to be a big deal over with the PSMoveService team as well. It would seem that these new controllers have a new PID as well as possible changes to the protocol.
Hopefully this is something that can be resolved, or I need to go and return the 3 pair of new controllers I just picked up... :(

nitsch, 2018-05-08:
These new controllers are currently not supported by psmoveapi.

And since you have the hardware at your hands: Why don't you try whether
changing the USB product ID works?

ChainmailBuddha, 2018-05-08:
Sadly, I only acquired the new controllers to work with JoustMania, and since they currently cannot do so, I have returned them and picked up some version 1 controllers which will.
However, I did want to make sure this was on the radar since I had not seen an issue raised yet for it.
I also know some other devs working on the issue now, so hopefully they can get some traction on it.

thp, 2018-06-01:
I have ordered one of these controllers and can report back once I receive those and have started playing around.

thp, 2018-06-01:
Also note that not all "PS4" PS Move controllers are actually Micro-USB controllers, some are still repackaged PS3 controllers (Mini-USB), e.g. the ones with "CECH-ZCM1E" as their part number.

As an example, this PS Move "PS4" Controller pack seems to be still the PS3 variant:

[image]

In other words:

 * Supported (same as PS3): CECH-ZCM1E
 * Not (yet) supported (Micro-USB): CECH-ZCM2J

thp, 2018-06-22:
I now have a CECH-ZCM2J. As a first quick test, updating the PID to the new value lets me set the controllers' color via USB + pair it. Bluetooth connection is unstable, but it might be due to low battery, let's see (currently charging it up, I just received and unpacked it..).

cboulay, 2018-06-22:
@thp Was the color setting done over USB?

nitsch, 2018-11-27:
Release 4.0.6 added basic support for CECH-ZCM2.

closed 2018-11-27 by nitsch