thp/psmoveapi #354: Windows builds issue

saadminhas, 2018-05-18:
Hi,

Downloaded the latest windows build 

**psmoveapi-4.0.5-windows-msvc2017-x64**

But as soon as I try to run psmove.exe or test_tracker.exe

It throws an error saying The program can't start because psmoveapi.dll is missing from your computer. Try reinstalling the program to fix this problem.

I can see the psmoveapi.dll is present in the lib folder.

Any ideas?

Kind Regards,
Saad

nitsch, 2018-05-18:
Add the path to psmoveapi DLLs to your system's DLL search path (i.e.
the PATH environment variable).  Alternatively, put the psmoveapi DLLs
into the same directory as the psmove executable you are trying to run.

thp, 2018-06-01:
Closing, as Alexander's comment should fix the issue.

closed 2018-06-01 by thp