thp/psmoveapi #355: Issue with running python scripts

rdFoe, 2018-05-30:
Hi,

I downloaded the latest version and built it for windows. When I want to run one of the python test scripts (for example always.py) I get the error "psmove modul not found". Did I miss something?

thp, 2018-06-01:
How did you build it? What does the `cmake` configure output show? You probably have to install Python and the [SWIG](http://www.swig.org/download.html) binary first before running cmake, so that it can pick up the path to those tools.

Alternatively, with the latest version, you can also just use the DLL + the `ctypes`-based [psmoveapi.py](https://github.com/thp/psmoveapi/blob/master/bindings/python/psmoveapi.py) without the need to build the bindings with SWIG. The API is a bit different, the old examples won't work, but there's a new example here: [new_api_events.py](https://github.com/thp/psmoveapi/blob/master/examples/python/new_api_events.py)

rdFoe, 2018-06-01:
Thanks for your quick response. I got it fixed! Had some errors with the SWIG bindings but got it fixed already.

thp, 2018-06-01:
Cool! Closing this issue. Feel free to open another one if you still got issues with the API.

closed 2018-06-01 by thp