thp/psmoveapi #358: Can't build under fedora

yeldarb1983, 2018-06-14:
[psmove api errors.txt](https://github.com/thp/psmoveapi/files/2100999/psmove.api.errors.txt)
[image]

nitsch, 2018-06-15:
What version of CMake is this?

nitsch, 2018-06-18:
I do not have any problem with CMake 3.11.2 on recent Debian and Ubuntu. But I noticed something else in your log: You did not check out the submodules. To do this, either clone wit `git clone --recursive ...` or run `git submodule init && git submodule update` inside the cloned repository to pull the submodules in. I am not exactly sure if this is related to CMake's "No SOURCES given to target" error but worth a try.

yeldarb1983, 2018-06-20:
Yeah, I'm kind of a babe-in-the-woods when it comes to git... I followed your instructions and got a bit further, though. I still only get to 55% before it errors out. This after updating and rebooting to make sure there wasn't something going on there. At least I can pair the controller and futz around with some of the settings now. That's a start at least. =)

thanks for your help so far, btw. I know dealing with newbs is a pain in the backside, so I really appreciate you taking the time.
[Compile 2.0.txt](https://github.com/thp/psmoveapi/files/2121302/Compile.2.0.txt)

nitsch, 2018-06-20:
It looks like a problem with the version of OpenCV you are using.
psmoveapi only works with OpenCV 2.x. Build scripts for Windows an Mac
properly retrieve OpenCV 2 but the actual version check in
src/tracker/CMakeLists.txt is missing. The line

  FIND_PACKAGE(OpenCV QUIET)

should rather be

  FIND_PACKAGE(OpenCV 2 QUIET)

If you already have OpenCV 2.x installed this should fix it. If you need
to build it yourself, have a look at scripts/macos/build-macos.

yeldarb1983, 2018-06-21:
Looks like it's compiling now, and it seems to work for the most part. I'm guessing the compiler is just being fussy about some minor syntax issues? 

At any rate, I got the api up and running, just need to figure out how to get the samples running, and maybe get it to detect the camera (ps3 eye before you ask).

[100% with warnings.txt](https://github.com/thp/psmoveapi/files/2122011/100.with.warnings.txt)

nitsch, 2018-06-21:
You are building without the tracker right now since no OpenCV could not
be found. If you want to use camera tracking, you need to build with the
tracker enabled.

yeldarb1983, 2018-06-21:
that's weird, I thought I had it opencv [sigh]. I take it I need to downgrade to opencv 2.0? I think I'm on 3 atm.

nitsch, 2018-06-21:
Yes, you will have to use OpenCV 2. Some Linux distributions have
dedicated packages for OpenCV 2, so you can install version 3 and 2 side
by side (i.e. no need for a downgrade). Not sure if Fedora does. But
building OpenCV yourself is not too hard either. Especially since
psmoveapi's build script for Mac OS basically contains everything you
need (see my previous posting).

nitsch, 2018-06-21:
BTW, you do not have to use OpenCV 2.0. Any 2.x is fine.

nitsch, 2018-07-08:
@yeldarb1983 Any updates on this?

yuripourre, 2018-08-27:
@yeldarb1983 @nitsch I could build in Fedora 27 using cmake version 3.11.2.

closed 2018-12-25 by nitsch