thp/psmoveapi #367: OpenCV 3 works! Ubuntu 18.04 "-DPSMOVE_BUILD_TRACKER=ON"

noabody, 2018-08-19:
Dependencies (based on install_dependencies.sh):
```shell
libudev-dev libbluetooth-dev libv4l-dev libopencv-dev openjdk-8-jdk ant liblwjgl-java python-dev mono-mcs swig3.0 freeglut3-dev python-sphinx python-pip
```
But needs a patch:
```diff
diff --git a/examples/c/distance_calibration.c b/examples/c/distance_calibration.c
index 14f8189..3c209c5 100644
--- a/examples/c/distance_calibration.c
+++ b/examples/c/distance_calibration.c
@@ -34,6 +34,7 @@
 
 #include "opencv2/core/core_c.h"
 #include "opencv2/highgui/highgui_c.h"
+#include "opencv2/imgcodecs/imgcodecs_c.h"
 
 #include "psmove.h"
 #include "psmove_tracker.h"
diff --git a/src/tracker/CMakeLists.txt b/src/tracker/CMakeLists.txt
index 606438a..c1e6d6b 100644
--- a/src/tracker/CMakeLists.txt
+++ b/src/tracker/CMakeLists.txt
@@ -28,7 +28,7 @@ ENDIF()
 
 # 1 - OpenCV
 IF(PSMOVE_BUILD_TRACKER)
-    FIND_PACKAGE(OpenCV 2 QUIET)
+    FIND_PACKAGE(OpenCV 3 QUIET)
     IF(OpenCV_FOUND)
         IF ("${OpenCV_LIBS}" STREQUAL "")
             message("OpenCV Libs was empty! Manually setting.")
diff --git a/src/tracker/camera_control.h b/src/tracker/camera_control.h
index 7900348..d086423 100644
--- a/src/tracker/camera_control.h
+++ b/src/tracker/camera_control.h
@@ -35,6 +35,7 @@ extern "C" {
 #include "psmove.h"
 #include "../psmove_private.h"
 #include "opencv2/core/core_c.h"
+#include "opencv2/videoio/videoio_c.h"
 
 struct _CameraControl;
 typedef struct _CameraControl CameraControl;
```

thp, 2018-08-27:
Cool! Wonder if those headers exist in OpenCV 2 as well, so we can support both versions easily?

nitsch, 2018-08-29:
Why are people so keen on dragging OpenCV 3 into this library? OpenCV 2
is well maintained. We are using its C API which is officially
deprecated in OpenCV 3.

noabody, 2018-08-29:
I don't have any opencv2 candidates available on Ubuntu 18.04.1.  Not an issue with the project, of course.  I just can't build without making those changes.  Would never suggest it as a solution but maybe save another Ubuntu user a bit of time.

```shell
libopencv-dev:
  Installed: 3.2.0+dfsg-4build2
  Candidate: 3.2.0+dfsg-4build2
  Version table:
 *** 3.2.0+dfsg-4build2 500
        500 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
        100 /var/lib/dpkg/status
```

thp, 2018-08-30:
I'll reopen this for now as we will likely have to eventually use OpenCV 3 on systems that don't ship with OpenCV 2 (of course, even on Ubuntu, there's still the possibility to build OpenCV 2 from source, but it's a bit more tedious than just installing the binaries + dev packages from the repos).

nitsch, 2018-08-30:
@noabody Is building OpenCV from source an option for you?

noabody, 2018-08-31:
Looked like it would overlap my package management OpenCV 3.2.  Was going to pull the sources from either debian sid or ubuntu xenial since that would create debs.  Anyhow, the patch I noted does build okay so I'm not sure it's worth trying to break my system.

But yes, I could probably build OpenCV2 from source.

closed 2020-01-04 by nitsch