thp/psmoveapi #370: Error when building on windows:

adangert, 2018-09-08:
when running:  .\scripts\visualc\build_msvc.bat 2017 x64

CMake Warning at cmake/OpenCVPackaging.cmake:23 (message):
  CPACK_PACKAGE_VERSION does not match version provided by version.hpp
  header!
Call Stack (most recent call first):
  CMakeLists.txt:1103 (include)

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
opencv_dep_CUDA_nppi_LIBRARY
    linked by target "opencv_core" in directory C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/modules/core
    linked by target "opencv_core" in directory C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/modules/core
    linked by target "opencv_core" in directory C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/modules/core
    linked by target "opencv_flann" in directory C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/modules/flann
    linked by target "opencv_imgproc" in directory C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/modules/imgproc
    linked by target "opencv_highgui" in directory C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/modules/highgui
    linked by target "opencv_features2d" in directory C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/modules/features2d
    linked by target "opencv_calib3d" in directory C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/modules/calib3d
    linked by target "opencv_ml" in directory C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/modules/ml
    linked by target "opencv_nonfree" in directory C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/modules/nonfree

-- Configuring incomplete, errors occurred!
See also "C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/build-x64/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Aaron/Documents/GitHub/JoustMania/psmoveapi/external/opencv/build-x64/CMakeFiles/CMakeError.log".
Failed to generate OpenCV solution

There was an error running one of the build commands. Please see the output for more information.

adangert, 2018-09-09:
had to add -DWITH_CUDA=OFF to build_msvc.bat

However now I'm getting the error:
LINK : fatal error LNK1104: cannot open file 'python37_d.lib' [C:\Users\Aaron\Documents\GitHub\JoustMania\psmoveapi\build-x64\_psmove.vcxproj]

adangert, 2018-09-09:
I had to download python with debug support for this to work.   -DWITH_CUDA=OFF still needs to be added to the bat file.

thp, 2018-09-11:
I guess you could disable linking OpenCV against Python to avoid having to install the debug version of Python?

closed 2020-09-05 by nitsch