thp/psmoveapi #371: psmove pair doesn't work with BlueSoleil Generic Bluetooth Driver

adangert, 2018-09-09:
This is what happens when trying to pair with a bluetooth adapter with the BlueSoleil driver on windows.

C:\Users\Aaron\Documents\GitHub\JoustMania\psmoveapi\build-x64>psmove pair
Connected controllers: 1
[PSMOVE WARNING] Magnetometer in e0:ae:5e:4e:7a:c5 not yet calibrated.
PSMove #1 connected via USB.
[PSMOVE WARNING] Failed to find a Bluetooth radioWARNING: Can't determine Bluetooth address.
Make sure Bluetooth is turned on.
[PSMOVE CRITICAL] Assertion fail in psmove_pair: host != NULL
Pairing of #1 failed.
Calibration data available and saved.

thp, 2018-09-11:
BlueSoleil isn't really supported, please use the Microsoft Bluetooth Stack. Note that you could try pairing with BlueSoleil as well, but you need to pass in the host Bluetooth address manually as parameter after the `pair` command (but nobody has tried, so I'm not sure if BlueSoleil will properly expose the controller as HID device).

adangert, 2018-09-12:
yeah doesn't seem to work either:

PS C:\Users\Aaron\Documents\GitHub\JoustMania\psmoveapi\build-x64> ./psmove pair 00:15:83:ef:21:55
Using user-supplied host address: 00:15:83:ef:21:55
Connected controllers: 1
[PSMOVE WARNING] Magnetometer in 00:06:f7:1b:dd:e2 not yet calibrated.
PSMove #1 connected via USB.
[PSMOVE WARNING] Failed to find a Bluetooth radioPairing of #1 failed.
Calibration data available and saved.

closed 2018-12-25 by nitsch