thp/psmoveapi #374: "Could not set DFU mode"

GAsplund, 2018-12-08:
Tried multiple versions, multiple USB ports. Does not work with neither BTDFU or USBDFU.
[image]

For everything I've trited to do, _psmove_set_operation_mode seems to be returning false in any case.

nitsch, 2018-12-08:
What version of Windows are you using?

What controller model are you using (see the string that begins with 
"CECH-" on the controller's back)?

GAsplund, 2018-12-08:
Win 10 ver. 1803
Controllers are CECH-ZCM1E

nitsch, 2018-12-08:
It's been quite a while but I just quickly looked into this:

Setting the DFU mode (as well as some other operations) is done using 
HID feature reports that the controller does not publicly advertise. The 
officially available HID reports are grouped into 3 different 
collections. Each of these shows up as a separate device on Windows. To 
use a specific HID report, you have to use the corresponding device 
(which is what we already do for reading/writing Bluetooth addresses and 
using other reports).

But I never got this to work with HID reports that weren't part of any 
collection; such as the ones for DFU mode and firmware info, for 
instance. I did not try on Windows 10 though. Maybe things work 
differently now. But it will be a while until I can set up a Windows 10 
test machine and try.

So, unfortunately, for now you have to use Linux if you really need the 
DFU feature. Alternatively, have a look at this:

 
https://github.com/nitsch/moveonpc/wiki/Hardware-and-firmware#user-content-enabling-usb-dfu-mode

It will allow you to switch to STDFU mode (but not to BTDFU) without 
software.

GAsplund, 2018-12-09:
I ran it in a VM, it works without any issue. Seems to be happening only under Windows.

closed 2020-09-05 by nitsch