thp/psmoveapi #375: Error in script usage when building

Derzo, 2018-12-24:
When I use the `call scripts/visualc/build_msvc.bat 2017 x64` command, I get this error. I also got an error about it not finding the vcvarsall.bat, but I managed to fix it by editing its path in the build_msvc.bat file. I tried googling this issue a while, but I couldn't get anything to work. I'm not that experienced with this stuff

```[ERROR:vcvarsall.bat] Error in script usage. The correct usage is:
Syntax:
    vcvarsall.bat [arch] [platform_type] [winsdk_version] [-vcvars_ver=vc_version] [-vcvars_spectre_libs=spectre_mode]
where :
    [arch]: x86 | amd64 | x86_amd64 | x86_arm | x86_arm64 | amd64_x86 | amd64_arm | amd64_arm64
    [platform_type]: {empty} | store | uwp
    [winsdk_version] : full Windows 10 SDK number (e.g. 10.0.10240.0) or "8.1" to use the Windows 8.1 SDK.
    [vc_version] : {none} for default VS 2017 VC++ compiler toolset |
                   "14.0" for VC++ 2015 Compiler Toolset |
                   "14.1x" for the latest 14.1x.yyyyy toolset installed (e.g. "14.11") |
                   "14.1x.yyyyy" for a specific full version number (e.g. 14.11.25503)
    [spectre_mode] : {none} for default VS 2017 libraries without spectre mitigations |
                     "spectre" for VS 2017 libraries with spectre mitigations

The store parameter sets environment variables to support Universal Windows Platform application
development and is an alias for 'uwp'.

For example:
    vcvarsall.bat x86_amd64
    vcvarsall.bat x86_amd64 10.0.10240.0
    vcvarsall.bat x86_arm uwp 10.0.10240.0
    vcvarsall.bat x86_arm onecore 10.0.10240.0 -vcvars_ver=14.0
    vcvarsall.bat x64 8.1
    vcvarsall.bat x64 store 8.1

Please make sure either Visual Studio or C++ Build SKU is installed.
Applying libusb dynamic CRT patch...

Building libusb
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: D:\Documents\psmoveapi\scripts\visualc\..\..\\external\libusb-1.0/msvc/libusb_static_2017.vcxproj
Failed to build libusb

There was an error running one of the build commands. Please see the output for more information```

nitsch, 2019-01-20:
@Derzo Did this solve your issue?

Derzo, 2019-09-01:
I fixed it, last time I tried I didn't realize that I have to recursively clone the repository and the troubles were coming from me trying to manually install all the libs

closed 2019-09-01 by Derzo