thp/psmoveapi #378: hid_get_feature_report fails sporadically on Windows 10

dquam, 2019-01-14:
hid_get_feature_report returns more bytes than requested. DeviceIoControl returns successfully but the results are not correct. ZCM2 models seem to require 20 bytes instead of 16 bytes. See also pull request #377.

dquam, 2019-01-14:
Looks like I'm not the first to discover this. [PSMoveService](https://github.com/cboulay/PSMoveService) seems to have encountered the [same thing](https://github.com/cboulay/PSMoveService/blob/edbb3141738387a88e4b89910c0883ebf5327f9e/src/psmoveservice/PSMoveController/PSMoveController.cpp#L63).

dquam, 2019-01-15:
Bluetooth. It seems to work correctly via USB. I also plan on testing on Windows 7 as soon as I can.

Yes, it seems to be every other attempt.

CECH-ZCM1U (no issues) and CECH-ZCM2E (problems).

Any of the examples or utilities using the PSMoveAPI exhibit the issue for me (e.g. battery or list). The symptom will be a crash in the PSMoveAPI constructor trying to instantiate the `std::string serial` or an assert fail in `psmove_get_serial`.

dquam, 2019-01-15:
I reproduced this on Windows 7 as well.

nitsch, 2019-01-20:
Phew ... after a lot more trial and error, I was finally able to establish a working Bluetooth connection with the ZCM2 on Windows 7. I was also able to reproduce the problem with the feature report and confirm that increasing the buffer size to at least 20 bytes fixes it.

Fixed in 57b87d5d9243e791f23bba39c05ad3ab587f5038.

Thanks!

closed 2019-01-20 by nitsch