thp/psmoveapi #381: [PSMOVE CRITICAL] Assertion fail in psmove_calibration_new: Unknown PS Move model

vasekhodina, 2019-02-13:
Hello,
It seems I cannot get past the pairing. Pairing works but other functions seem to fail with the error:
[PSMOVE CRITICAL] Assertion fail in psmove_calibration_new: Unknown PS Move model

I'm running fedora 28, with bluez version bluez-5.50.

Below is the command line log, some commands didn't work because those weren't run with sudo, but I'm leaving them for completeness sake. Please let me know 
```
$ sudo ./psmove pair
Connected controllers: 1
[PSMOVE CRITICAL] Assertion fail in psmove_calibration_new: Unknown PS Move model
PSMove #1 connected via USB.
[PAIRING LINUX] Found host adapter: f4:96:34:11:b4:2f (name=hci0)
[PAIRING LINUX] File /var/lib/bluetooth/F4:96:34:11:B4:2F/A4:53:85:D5:00:73/info needs updating.
[PAIRING LINUX] Running: 'systemctl stop bluetooth.service'
[PAIRING LINUX] Writing file: /var/lib/bluetooth/F4:96:34:11:B4:2F/A4:53:85:D5:00:73/info
[PAIRING LINUX] File /var/lib/bluetooth/F4:96:34:11:B4:2F/cache/A4:53:85:D5:00:73 is already up to date.
[PAIRING LINUX] Running: 'systemctl start bluetooth.service'
Pairing of #1 succeeded!
Controller address: a4:53:85:d5:00:73
Calibration data available and saved.
$ ./psmove list
[PAIRING LINUX] Found host adapter: f4:96:34:11:b4:2f (name=hci0)
Host address: f4:96:34:11:b4:2f
[PSMOVE CRITICAL] Assertion fail in psmove_get_serial: move != NULL
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Aborted (core dumped)
$ ./psmove --help
Command not found: '--help', use 'help' to see a list of commands.
$ ./psmove help
Usage: ./psmove <cmd>

Where <cmd> is one of:

    pair                 ... Pair connected USB controllers to a host
    remote-pair          ... Pair connected USB controller to a moved host
    daemon               ... Serve locally-connected controllers via UDP
    register             ... Register already-paired controllers
    calibrate            ... Calibrate magnetometers of controllers
    battery              ... Visualize the battery charge on connected controllers
    extensions           ... Show sharp shooter and racing wheel extension data
    list                 ... List connected controllers

  Debugging Tools (for developers):

    dump-calibration     ... Show the stored calibration information
    auth-response        ... Challenge-response authentication dev tool
    dfu-mode             ... Switch to DFU mode (potentially dangerous)
    firmware-info        ... Show information about the controller firmware

  Test Tools:

    responsiveness       ... Test how quickly the controllers react
    led-pwm-frequency    ... Test LED PWM frequency modulation

$ ./psmove register
This program must be run as root (or use sudo).
$ sudo ./psmove register
Usage: register [--ps4] bluetooth-address
No Bluetooth address supplied.
$ sudo ./psmove list
[PAIRING LINUX] Found host adapter: f4:96:34:11:b4:2f (name=hci0)
Host address: f4:96:34:11:b4:2f
[PSMOVE CRITICAL] Assertion fail in psmove_calibration_new: Unknown PS Move model
Controller 0: a4:53:85:d5:00:73 (USB, battery: charging)
$ sudo ./example
Connected controllers: 1
[PSMOVE CRITICAL] Assertion fail in psmove_calibration_new: Unknown PS Move model
Serial: a4:53:85:d5:00:73
Connected via USB.
```
If I run example_new_api there is output, but all values are 0.
```
Accel: 0.00, 0.00, 0.00
Gyro: 0.00, 0.00, 0.00
Magnetometer: 0.00, 0.00, 0.00
Connection [x] USB [ ] Bluetooth; Buttons: 0x0000
```

nitsch, 2019-02-13:
It is complaining about an unknown PS Move model. Which one are you 
using, ZCM1 or ZCM2? Anything special about it?

Which version of psmoveapi are you using, which commit exactly?

vasekhodina, 2019-02-20:
ZCM2, I knew I forgot to add some essential information to the message.
I'm using version 4.0.7 latest commit 99f1f9f1c1eaa557c5d346309a5b1d75f58dda42

nitsch, 2019-02-20:
These issues are already fixed in the current master branch. You can 
build from source to give it a try.

vasekhodina, 2019-03-13:
Hello,
sorry for the long time without a response. It works with the newest commit. Thank you for your help.

closed 2019-03-13 by vasekhodina