thp/psmoveapi #382: Request: for Magnometter, need separe 2 psmove with same PID

Zangetsu38, 2019-05-16:
so, issue is for psmove ps4 without magnometter, is Model_ZCM1U for me, same VID/PID,  just get difference reference with 460 and no 400 for real PS3 psmove with Magnometer

nitsch, 2019-05-16:
> so, issue is for psmove ps4 without magnometter, is Model_ZCM1U for
> me, same VID/PID,  just get difference reference with 460 and no 400
> for real PS3 psmove with Magnometer

What is 460 and 400? Are these PS3 or PS4 firmware versions?

> Here is probleme because it not have magnometter but psmoveapi think 
> it have that, and that causing probleme for all, no can works 
> correctly

Are you talking about psmoveapi's tracking assuming the presence of a 
magnetometer for all ZCM1 models? If so, you can use 
psmove_set_orientation_fusion_type() to set the algorithm that computes 
controller orientation from the raw sensor data. Currently only 
OrientationFusion_MadgwickIMU works without using the magnetometers. 
This is also the default algorithm automatically chosen for ZCM2 models.

If you are talking about something else, please describe your problem 
more precisely. I have great trouble understanding anything you say.

Zangetsu38, 2019-05-16:
I said just my PSMove not have Megnometter, and is ZCM1 model for psmoveapi, because exept write on psmove ZCM1U and ref 460 inside usb connect(after vid and pid), not have other difference.
But so lik,e i have said, if i calibrate, and after delete magnometer file, that works normally
so for that i said need add one think for psmoveapi know and understund both is separed and not same controller, ZCM1U, is first ps4 psmove out