thp/psmoveapi #383: Mingw cross compile not generating java bindings

rafaellago, 2019-05-16:
When I cross compile using mingw64 the java binding files are not generated. I'm using a codenvy instance to build using the buildscript provided. 

I had to revert bindings/CMakeLists.txt to a previous version because swig_add_library() was causing an error that was not letting build anything. After the change back to swig_add_module() it built ok. 

Attached are the CMake{Output,Error}.log 
[CMakeError.log](https://github.com/thp/psmoveapi/files/3188731/CMakeError.log)
[CMakeOutput.log](https://github.com/thp/psmoveapi/files/3188732/CMakeOutput.log)

I woldn't mind to have the files provided together with the release packages, just as it happens with the python bindings.

nitsch, 2019-05-16:
> When I cross compile using mingw64 the java binding files are not
> generated. I'm using a codenvy instance to build using the
> buildscript provided.
I assume the script you are talking about is scripts/build_package.sh 
with BUILD_TYPE set to linux-cross-mingw64.

The logs you attached do not show the CMake configure step that would be 
helpful for inspecting your psmoveapi build configuration. For instance, 
have a look at the output for the latest CI-build using mingw64:

https://travis-ci.org/thp/psmoveapi/jobs/496289657#L8160-L8215

The summary of bindings to be built can be found starting on line 8207. 
In this particular instance, Java is not installed on the build machine, 
that is why it says "JNI not found".

Have a look at your output of CMake's initial configure step for 
psmoveapi to find out what it says about Java bindings on your machine. 
That should clear things up.

rafaellago, 2019-05-16:
> I assume the script you are talking about is scripts/build_package.sh with BUILD_TYPE set to linux-cross-mingw64.

Yes, that's correct. 

And according to the output ([build_mingw64.log](https://github.com/thp/psmoveapi/files/3189051/build_mingw64.log) line 6451), it hasn't built for the same reason "JNI not found". Which is weird, since I use this  instance to build my java projects and the Linux build ([build_linux.txt](https://github.com/thp/psmoveapi/files/3189056/build_linux.txt)) generates the java bindings without problems.

```
user@b609a44c40a4:/projects/psmoveapi/scripts$ java -version
openjdk version "1.8.0_171"
OpenJDK Runtime Environment (build 1.8.0_171-8u171-b11-0ubuntu0.16.04.1-b11)
OpenJDK 64-Bit Server VM (build 25.171-b11, mixed mode)
```

rafaellago, 2019-05-16:
Just an update. 

2 things I saw, the windows build didn't set the JAVA_HOME/bin to the path, although I don't think this causes the problem since it was already set by default. 

The second thing, I edited the bindings/CMakeLists.txt  and removed the QUIET option when it searches for JNI. And then I got the following message when trying to build: -- Could NOT find JNI (missing:  JAVA_AWT_LIBRARY JAVA_JVM_LIBRARY) 

Since this is a headless environment it does make sense to not have those libraries installed, I installed the non headless version of openjdk but had the same error.

I'll try again at home with my linux VM which has a desktop environment.

Thanks for the pointer at where to start searching for the problem.

rafaellago, 2019-05-17:
Another update. I was able to generate the .jar and .dll but had to make some changes in some files. 
cmake/x86_64-w64-mingw32.toolchain i had to add the following lines
```
SET(JAVA_JVM_LIBRARY /usr/lib/jvm/java-8-openjdk-amd64/include)
SET(JAVA_AWT_LIBRARY /usr/lib/jvm/java-8-openjdk-amd64/include)
```
And because cmake was thinking that my environment was windows, I had make this change on bindings/CMakeLists.txt  on top of the mentioned on the OP. 

```
                    # COMMAND ${CMAKE_CURRENT_LIST_DIR}/copy_java.bat ${PSMOVEAPI_JAVA_PKG_DIR}/
                    COMMAND cp *.java ${PSMOVEAPI_JAVA_PKG_DIR}/
```

With those changes I was able to generate the files on the build dir. For some reason it hasn't packaged in the final zip file. But I can work with what I have, for now. 

I'm not sure if this is enough to close the issue

nitsch, 2019-05-19:
> cmake/x86_64-w64-mingw32.toolchain i had to add the following lines
> ```
> SET(JAVA_JVM_LIBRARY /usr/lib/jvm/java-8-openjdk-amd64/include)
> SET(JAVA_AWT_LIBRARY /usr/lib/jvm/java-8-openjdk-amd64/include)
> ```

I would have expected CMake's module for finding JNI to properly locate 
these libs. Maybe it tries locating them on a Windows system since you 
are cross-compiling?

> I'm not sure if this is enough to close the issue

This seems rather messy and should be fixed, so I suggest you leave this 
ticket open as a reminder.

Thanks for taking the time to investigate.

rafaellago, 2019-05-20:
> Maybe it tries locating them on a Windows system since you are cross-compiling?

Yes, maybe. I think that the cross compile sets an incorrect "Windows" flag somewher, because the CmakeLists from the bindings, use the Windows block, which calls the copy_java.bat. 

Another issue I found when testing, is that it puts the wrong library name on the JNI call inside the generated java interface. And it misses 2 other necessary dlls. After changing the psmove.i file to correct those problems, it ran without any issues. But I need to test more to see if all is really ok. 

If I figure out how to fix this without breaking anything I'll do a PR.

commit 74a65117 (2020-06-09): Merge pull request #408 from rafaellago/master

Fixes to Java and Csharp bindings (Fixes #383 and possibly #341)
closed 2020-06-09 by thp