thp/psmoveapi #384: move.get_serial() breaks when in thread

adangert, 2019-05-21:
This was run on OSX.

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/aaron/git/JoustMania/controller_process.py", line 10, in main_track_move
    move = common.get_move(move_serial, move_num)
  File "/Users/aaron/git/JoustMania/common.py", line 47, in get_move
    print(move.get_serial())
  File "/Users/aaron/git/psmoveapi/build/psmove.py", line 453, in get_serial
    return _psmove.PSMove_get_serial(self)
TypeError: in method 'PSMove_get_serial', argument 1 of type 'PSMove *'

thp, 2019-05-21:
Can you create a [Short, Self Contained, Correct Example](http://sscce.org) that demonstrates that issue? In general, PS Move API isn't threadsafe (so use a mutex when accessing any PS Move functions/attributes, so that only one thread accesses it at once), although it looks like this issue might not be threading related, hence SSCCE.

adangert, 2019-05-21:
hmm, yeah I'll look into it, in my simple example it looked like threading did work. I'll keep running though the code to see where the issue might be.

adangert, 2019-05-21:
This is what breaks it:
```
import psmove
import threading
import controller_process
def test():
    print("starting")
    move = psmove.PSMove(0)
    print(move.get_serial())

if __name__ == "__main__":
    moves = [psmove.PSMove(x) for x in range(psmove.count_connected())]
    t1 = threading.Thread(target=controller_process.main_track_move2, args=())
    t1.start()
```

adangert, 2019-05-21:
This seems to break it as well:

```
import psmove
import threading

if __name__ == "__main__":
    moves = [psmove.PSMove(x) for x in range(psmove.count_connected())]
    move = psmove.PSMove(0)
    print(move.get_serial())
```

adangert, 2019-05-22:
Just tried, the above code works on Windows, not OSX however.

thp, 2019-06-04:
Then it's because you are trying to open a controller two times, and why this might happily work on Windows, it apparently doesn't work on macOS.

You should only open each handle once (e.g. `psmove.PSMove(0)` only once), and re-use the handles if you want to create them from multiple places.

Maybe we should add something like this to be detected at least...

nitsch, 2020-01-04:
@adangert Did you try the suggested solution? Does it fix your problem?

adangert, 2020-01-04:
I think that solution works, I just haven't had the time to refactor the joustmania codebase to incorporate it.

closed 2020-01-05 by nitsch