thp/psmoveapi #386: Windows 10 1903 version breaks ps move connectivity

supersmo, 2019-07-10:
After updating to [Windows 10 version 1903](https://support.microsoft.com/en-us/help/4498140) the psmove controller disconnects after a second or so and the LED doesn't stop blinking. The update also updates the Microsoft bluetooth driver to newer versions even if the driver date is still 2006-06-21.

**Game devices in the control panel**
When connecting I see the Motion Controller show up for a few seconds and then disappear in the game devices app in the control panel (a.k.a. joy.cpl).

**Blue tooth devices**
When I check the bluetooth devices it says that the controller is paired and when I press the ps button on the move controller it also says connected for a few seconds before disconnecting.

**Event Viewer**
In the "Event viewer -> WIndows logs -> System"
I see entries of failed connects.
Here is one sample from the log: (OS is in Swedish)
The error message is translated to English is: "The first connection to Bluetooth HID-device failed. The device has been removed as a personal or shared device. You need to reinstall the device."
```
Loggnamn:      System
Källa:         HidBth
Datum:         2019-07-10 23:23:04
Händelse-ID:   4
Aktivitetskategori:Ingen
Nivå:          Fel
Nyckelord:     Klassiskt
Användare:     Saknas
Dator:         DESKTOP-AVVQ7UQ
Beskrivning:
Första anslutningen till Bluetooth HID-enhet  misslyckades. Enheten har tagits bort som personlig eller delad enhet. Du måste installera om enheten.
Händelsens XML-data:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="HidBth" />
    <EventID Qualifiers="49153">4</EventID>
    <Level>2</Level>
    <Task>0</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2019-07-10T21:23:04.703512800Z" />
    <EventRecordID>647</EventRecordID>
    <Channel>System</Channel>
    <Computer>DESKTOP-AVVQ7UQ</Computer>
    <Security />
  </System>
  <EventData>
    <Data>\Device\_HID00000019</Data>
    <Data>
    </Data>
    <Binary>000000000200500000000000040001C0000000000000000000000000000000000000000000000000</Binary>
  </EventData>
</Event>
```

So what makes me believe it is the Windows 1903 that causes the issue?

I've tried to get psmoveapi to work on 2 computers. My desktop computer which already had the 1903 version and a laptop with an older version (I think it was v1803.) The PS Move controllers would connect properly on the laptop but not on the desktop even though I was using the same bluetooth dongle and using Microsoft's drivers on both computers.

The only difference I noticed was that the driver versions differed and seemed to be related to the Windows version and I couldn't find an easy way to downgrade the drivers on the desktop computer, so after troubleshooting for hours I decided to update the laptop to Windows 1903 to see what would happen and after the update the laptop had the same immediate disconnect behavior. I've also tried the laptop's built-in bluetooth which was also working prior to the update but I get the same behavior using that.

I'll now try reverting the Windows 1903 update on the laptop. (I couldn't do that on the desktop since it was more than 10 days since the update had been applied so that convenient system restore point seems to be gone.)

supersmo, 2019-07-10:
I've reverted back to Windows 10 1809 version on the laptop and the psmove controllers are now able to connect again.

adangert, 2019-07-13:
Can confirm controllers for me no longer initially connect (however if the controller was synced before 1903, then they will continue to work, it's only new controllers trying to be paired)

supersmo, 2019-07-16:
I found an update in the 1809 update that looked like it could be related to the connections issues:

[KB4503293](https://support.microsoft.com/sv-se/help/4503293/windows-10-update-kb4503293):
"Addresses a security vulnerability by **intentionally preventing connections between Windows and Bluetooth devices** that are not secure and use well-known keys to encrypt connections, including security fobs. If BTHUSB Event 22 in the Event Viewer states, “Your Bluetooth device attempted to establish a debug connection….”, then your system is affected. Contact your Bluetooth device manufacturer to determine if a device update exists. For more information, see [CVE-2019-2102](https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2019-2102) and [KB4507623](https://support.microsoft.com/en-us/help/4507623)."

I haven't seen the "BTHUSB Event 22" in the Event Viewer but I thought it was worth trying to remove to try it out so I uninstalled "Update for Microsoft Windows KB4503293", but I had to uninstall KB4507453 first in order to see it in the list of updates to uninstall.

But the issue still remains. :-(

supersmo, 2019-08-06:
Found a pull request in the psmoveservice project. https://github.com/psmoveservice/PSMoveService/pull/634
Haven't verified if the pairing works with the fix yet, but if it works, the same kind of changes might help fix psmoveapi.

adangert, 2019-08-06:
That probably would fix the problem!

adangert, 2019-08-26:
Looks like the problem is solved, I've updated the joustmania windows build on steam, Although now running Joustmania as an Administrator is required on windows.

closed 2019-08-26 by thp