thp/psmoveapi #392: test_tracker.exe takes forever

oscaramos, 2019-09-16:
I'm using the new "psmoveapi-4.0.8-mingw32.zip" and Cl Eye Driver. I've paired and calibrated my PS Move. 
When I execute test_tracker.exe, it keeps running forever, also takes 20% of CPU all the time.

```
D:\IHC\psmoveapi-4.0.8-mingw32\bin>test_tracker.exe
### Found 1 controllers.
Trying to init PSMoveTracker...Warning: No lens calibration files found.
```

nitsch, 2019-09-25:
Did it behave differently in the previous releases?

oscaramos, 2019-09-25:
It's ok with "psmoveapi-4.0.7-mingw32"

```
### Found 1 controllers.
Trying to init PSMoveTracker...Warning: No lens calibration files found.
C:\Users\Oscar\AppData\Roaming\.psmoveapi\colormapping.dat is too old - not restoring colors.
OK
Opening controller 0
Calibrating controller 0...color mappings saved.
OK, auto_update_leds is enabled
x:     119.50, y:     119.50, r:     167.58
x:     119.50, y:     119.50, r:     167.58
x:     119.50, y:     119.50, r:     167.58
x:     119.50, y:     119.50, r:     167.58
```

nitsch, 2019-09-25:
Can you give a previous build a try? This one does not yet include the
switch to OpenCV 3.

Go to https://ci.appveyor.com/project/thp/psmoveapi/builds/26966256 and
choose either the x64 or the x86 build, then select the "Artifacts" tab
and download the .zip file that contains the binaries.

oscaramos, 2019-09-26:
This shows:
```
D:\psmoveapi-4.0.7-16-g59d998f-windows-msvc2017-x86\bin>test_tracker.exe
### Found 1 controllers.
Trying to init PSMoveTracker...Could not init PSMoveTracker.
```

nitsch, 2019-09-28:
Is there a particular reason why you were using the MinGW release builds?

Can you try the psmoveapi-4.0.7-msvc2017 release? There should be no
difference, but just to make sure.

oscaramos, 2019-09-30:
I tried "psmoveapi-4.0.8-windows-msvc2017-x86".
```
D:\proyects\psmoveapi-4.0.8-windows-msvc2017-x86\bin>test_tracker.exe
### Found 1 controllers.
Trying to init PSMoveTracker...Warning: No lens calibration files found.
C:\Users\Oscar\AppData\Roaming\.psmoveapi\colormapping.dat is too old - not restoring colors.
OK
Opening controller 0
Calibrating controller 0...OpenCV: terminate handler is called! The last OpenCV error is:
OpenCV(3.4.8-pre) Error: Bad argument (Array should be CvMat or IplImage) in cvGetSize, file C:\projects\psmoveapi\external\opencv\modules\core\src\array.cpp, line 1246
```
After this I get "test_tracker.exe has stopped working".

I also tried "psmoveapi-4.0.8-windows-msvc2017-x64", it shows me the exact output.

oscaramos, 2019-09-30:
"Is there a particular reason why you were using the MinGW release builds?"
Well, no. I can use the microsoft version

nitsch, 2019-09-30:
Sorry, I was not clear enough. I meant the 4.0.7 release for MSVC. To
see if that one also works or if it is (for some reason) a MinGW vs.
MSVC issue.

oscaramos, 2019-10-03:
It works in 4.0.7 x86 release for MSVC:
```
D:\proyects\psmoveapi-4.0.7-windows-msvc2017-x86\bin>test_tracker.exe
### Found 1 controllers.
Trying to init PSMoveTracker...Warning: No lens calibration files found.
C:\Users\Oscar\AppData\Roaming\.psmoveapi\colormapping.dat is too old - not restoring colors.
OK
Opening controller 0
Calibrating controller 0...ERROR - retrying
Calibrating controller 0...ERROR - retrying
Calibrating controller 0...color mappings saved.
OK, auto_update_leds is enabled
x:     186.00, y:     168.00, r:      12.00
x:     186.00, y:     168.00, r:      12.00
x:     305.00, y:     201.50, r:      16.50
x:     275.00, y:     201.00, r:      34.00
x:     275.00, y:     201.00, r:      34.09
```
Also it appear a windows that tracks the ps move.
The x64 version has the same behavior.

nitsch, 2019-10-05:
I was finally able to test test_tracker.exe on a Windows machine today.

- Windows 10
- MSVC 2017 x64 builds
- CL Eye driver 5.3.0.0341
- PlayStation Eye camera

I could, however, not completely reproduce the behaviour you described:

The 4.0.7 release ran just fine. So did the version from the CI server
that I asked you to try. The 4.0.8 release has the same OpenCV issue, so
this is definitely busted.

nitsch, 2020-01-18:
It seems as if that particular version of OpenCV used in the psmoveapi 4.0.8 release build is the problem. The current ones work just fine while rolling back to that exact commit breaks the tracker.

@oscaramos Could you go to https://ci.appveyor.com/project/thp/psmoveapi/build/job/7tpcdlchndpkbb7u/artifacts, download the .zip file and confirm this version is working for you, too?

closed 2020-09-24 by nitsch