thp/psmoveapi #395: Cannot open move for retrieving serial - Handler callbacks not being called

LegendL3n, 2019-12-04:
Hi there!

I'm trying to register my callback and have it called to print the serial, however it's giving me these errors:

> [PSMOVE WARNING] Failed to connect to controller #0
[PSMOVE CRITICAL] Assertion fail in on_monitor_event: Cannot open move for retrieving serial!

This is when I connect the move, I'm also at the same time running
> sudo ./psmove daemon

My code pretty simple, as follows:

```
#include <iostream>
#include "../include/psmoveapi.h"

using namespace psmoveapi;

class LEDHandler : public Handler {
public:
    // nothing to clean up
    ~LEDHandler() {
        std::cout << "Hmm.. Was this expected?" << std::endl;
    }

    void connect(Controller *controller) {
        std::cout << "Connect" << std::endl;
    }

    void update(Controller *controller) {
        std::cout << "Update " << controller->serial << std::endl;
    }

    void disconnect(Controller *controller) {
        std::cout << "Disconnect" << std::endl;
    }
};

int main() {
    std::cout << "Hi" << std::endl;

    LEDHandler* myHandler = new LEDHandler();
    PSMoveAPI move(myHandler);

    while(true) {
        move.update();
    }
}
```

I'm connect it via both USB and Bluetooth and the output of the daemon is just
>0 connected
1 connected
2 connected

thp, 2019-12-04:
Try NOT running the daemon and either making sure the udev rule is
installed, the hidraw devices are chmod'ed 0666 or you are running your
program as root.

The daemon mode is usually only needed when writing networked applications
and can interfere with running another PS Move API using program at the
same time.

Hope this helps,
Thomas

LegendL3n, 2019-12-06:
I can't believe my problem was not running as root...

Thank you so much

closed 2019-12-08 by nitsch