thp/psmoveapi #396: failed to build libusb

ychdxx, 2019-12-08:
Apply
[image]
please mailme 34120257@QQ.com. thanks, I hope you to help me.

nitsch, 2019-12-08:
Do the .vcxproj files mentioned in the error message exist in your `external/libusb-1.0/msvc/` subdirectory?

If they don't, you probably did not clone the repository using the `--recursive` option for fetching submodules. You can fix this by running `git submodule init && git submodule update` in your local clone.

ychdxx, 2019-12-10:
I get,thks,myfriend, you are greate.

 
If they don't, you probably did not clone the repository using the --recursive option for fetching submodules. You can fix this by running git submodule init &amp;&amp; git submodule update in your local clone.

ychdxx, 2019-12-11:
Hi,my friend,Thanks to your help,but when I reference psmoveapi.dll into c#,there is a error  "could not be added,please make sure tha the file is accessible,and that it is a valid assembly or COM component".I need use psmove in unity,best unity4.6. What should I do?.

closed 2020-01-04 by nitsch