thp/psmoveapi #397: How using io.thp.psmove;

ychdxx, 2019-12-12:
my psmove's red led  eventually remain lit .so I reference psmoveapi.dll into c#,there is a error "could not be added,please make sure tha the file is accessible,and that it is a valid assembly or COM component".I need use psmove in unity,best unity4.6. What should I do?

nitsch, 2019-12-13:
The psmoveapi.dll is not a C# assembly, it is what C# refers to as
"native code". You have to either DllImport and wrap this native library
yourself or use the C# bindings that psmoveapi generates using SWIG.

Check the build log, find the section "Language bindings" and check if
C# bindings are listed as active in the build.

ychdxx, 2019-12-13:
I downloaded UniMove ,    [DllImport("libpsmoveapi")]
    private static extern int psmove_count_connected();this return 0.but my psmove's red led eventually remain lit

ychdxx, 2019-12-14:
[DllImport("psmoveapi")]
    private static extern int psmove_count_connected(); DllNotFoundException: E:/test/psmove/UniMove-master4.3/unitypsmove4.6/Assets/Plugins/psmoveapi.dll

ychdxx, 2019-12-14:
"Language bindings" is default ,I no change everything

closed 2020-09-05 by nitsch