thp/psmoveapi #398: I can only one psmove

mjsf12, 2019-12-13:
if i power off first one , i can pair second one. But, not the two at the same time.
`
C:\Users\Marcilio Junior\Downloads\psmoveapi-4.0.8-mingw64\bin>psmove.exe pair
Connected controllers: 1
[PSMOVE WARNING] Magnetometer in 00:06:f7:1f:7e:d4 not yet calibrated.
PSMove #1 connected via USB.

    Unplug the controller.

    Now press the controller's PS button. The red status LED
    will start blinking. Whenever it goes off, press the
    PS button again. Repeat this until the status LED finally
    remains lit. Press Ctrl+C to cancel anytime.
Connection verified.
Pairing of #1 succeeded!
Controller address: 00:06:f7:1f:7e:d4
Calibration data available and saved.

C:\Users\Marcilio Junior\Downloads\psmoveapi-4.0.8-mingw64\bin>psmove.exe pair
Connected controllers: 2
[PSMOVE WARNING] Magnetometer in 00:07:04:ac:40:b3 not yet calibrated.
PSMove #1 connected via USB.

    Unplug the controller.

    Now press the controller's PS button. The red status LED
    will start blinking. Whenever it goes off, press the
    PS button again. Repeat this until the status LED finally
    remains lit. Press Ctrl+C to cancel anytime.
Connection verified.
Pairing of #1 succeeded!
Controller address: 00:07:04:ac:40:b3
Calibration data available and saved.
Error connecting to PSMove #2
`
*  Poweroff the first one.
`
C:\Users\Marcilio Junior\Downloads\psmoveapi-4.0.8-mingw64\bin>psmove.exe pair
Connected controllers: 1
[PSMOVE WARNING] Magnetometer in 00:07:04:ac:40:b3 not yet calibrated.
PSMove #1 connected via USB.

    Unplug the controller.

    Now press the controller's PS button. The red status LED
    will start blinking. Whenever it goes off, press the
    PS button again. Repeat this until the status LED finally
    remains lit. Press Ctrl+C to cancel anytime.
Connection verified.
Pairing of #1 succeeded!
Controller address: 00:07:04:ac:40:b3
Calibration data available and saved.

C:\Users\Marcilio Junior\Downloads\psmoveapi-4.0.8-mingw64\bin>
`
System :
* windows 10 1909
* Chinese CRS 4.0 Dungle
* 2 CECH-ZCM1M (PS MOVE)
* psmoveapi-4.0.8-mingw64 (msvc2017 version same problem )

it's a bug or problem in my hardware (Psmove or USB dungle)?
how i get log?
English isn’t my first language, so please excuse any mistakes.
I’m working on improving my English. :D

nitsch, 2019-12-13:
Please describe in more detail how you are handling the controllers
during the steps described above. When pairing the controllers, do you
actually unplug them?

What logs do you want to get?

Besides, if pairing worked initially (and you did not change the
Bluetooth radio or paired the controller to another radio in the
meantime), you do not need to run "psmove.exe pair" again later to
establish a Bluetooth connection for this controller. Simply press the
controller's PS button to do that. Pairing is required only once per
controller to introduce controller and operating system to one another,
so to speak.

mjsf12, 2019-12-14:
> What logs do you want to get?
anyone,i just want to know what's happening.
 

> Please describe in more detail how you are handling the controllers
> during the steps described above. When pairing the controllers, do you
> actually unplug them?

yes, i unplug them.

the problem is a i can't use both , if i establish a Bluetooth connection when a first , the second no establish a connection,While i don't power off the fist one, i cant.

after i pair both , i press PS button , but only one  the light it's on, the second just blink and then turns off.

But, i think this problem is my USB dongle, it not connect 2 devices together, i think, i will test with another and close this issues if it is that.

mjsf12, 2019-12-14:
it's a Dongle problem. sorry about that.

closed 2019-12-14 by mjsf12

Thorvi, 2021-09-20:
Hello! did you get any results? I have the same problem

thp, 2021-09-23:
Try a different Bluetooth adapter. For example, @adangert recommends a "Class 1, Bluetooth 4.0 Adapter" in the [JoustMania Hardware Documentation](https://github.com/adangert/JoustMania#hardware), and that surely works for multiple controllers. Also, if you are on Windows, check out the [Windows Pairing Troubleshooting](https://psmoveapi.readthedocs.io/en/latest/pairing.html#windows) docs.