thp/psmoveapi #400: ps4 psmove pairing not working on linux (raspberry pi)

adangert, 2020-01-22:
when attempting to pair with sudo ./psmove pair the following prompt comes up:

    Press the controller's PS button. The red status LED
    will start blinking. This might take a few seconds. If
    the connection is established, the status LED will
    finally remain lit. Press Ctrl+C to cancel anytime.

However even after the controller connects successfully to the pi the prompt never goes away until you hit Ctrl+C, previously the script checked if the controller was connected after a couple of second and then went away on it's own. This only happens with the ps4 move controller and not the ps3 one on linux.

nitsch, 2020-01-22:
> previously the script checked if the controller was connected after a
> couple of second and then went away on it's own.

Previously this code for the ZCM2 did not even exist. ;-)

Can you build with debug output and check what it has to say? The code
is repeatedly checking for a successful Bluetooth connection and will
leave the loop if it detects one. This works for me, but this has not
been widely tested. Especially not on a Raspberry Pi.

adangert, 2020-01-22:
Oh I mean I've tested the ps4 controllers successfully on the pi previously. So something has changed in the past few of months.

nitsch, 2020-01-22:
> Oh I mean I've tested the ps4 controllers successfully on the pi 
> previously. So something has changed in the past few of months.

Yes, there was a Python script for setting up these controllers. I added
the functionality to `psmove pair` for usability. But that Python script
had to be aborted, too. It did not check the Bluetooth connection.

adangert, 2020-01-26:
This is currently what the output of the file looks like:

```pi@raspberrypi:~/psmoveapi/build $ sudo ./psmove pair
Connected controllers: 1
PSMove #1 connected via USB.
[PAIRING LINUX] Found host adapter: dc:a6:32:07:3b:bd (name=hci0)
[PAIRING LINUX] File /var/lib/bluetooth/DC:A6:32:07:3B:BD/DC:AF:68:EE:EC:3E/info needs updating.
[PAIRING LINUX] Running: 'systemctl stop bluetooth.service'
[PAIRING LINUX] Writing file: /var/lib/bluetooth/DC:A6:32:07:3B:BD/DC:AF:68:EE:EC:3E/info
[PAIRING LINUX] File /var/lib/bluetooth/DC:A6:32:07:3B:BD/cache/DC:AF:68:EE:EC:3E is already up to date.
[PAIRING LINUX] Running: 'systemctl start bluetooth.service'
[PAIRING LINUX] Registering PIN code agent

    Press the controller's PS button. The red status LED
    will start blinking. This might take a few seconds. If
    the connection is established, the status LED will
    finally remain lit. Press Ctrl+C to cancel anytime.
^C[PAIRING LINUX] Unregistering PIN code agent
Pairing of #1 succeeded!
ioctl (GFEATURE): No such device
Controller address: (null)
Calibration data available and saved.
```

I have the output of make -j4 -d, but it's quite long

adangert, 2020-01-26:
After doing some debugging it looks like it just gets stuck in this loop: https://github.com/thp/psmoveapi/blob/c0e5861d86fd4a4decd8e75c58b0ff61ca73ec0d/src/platform/psmove_port_linux.cpp#L542
even if you disconnect the controller or push the playstation sync button.

nitsch, 2020-01-26:
> it looks as if dbus_device_id is never set

That means the controller never requested a PIN code. Which also matches
the absence of any debug output between registering and unregistering
the PIN agent.

Did I understand correctly that the only problem is `psmove pair` not
exiting automatically while the Bluetooth connection with this
controller is working properly after the pairing procedure anyway?

How did you set up your ZCM2 controllers for Bluetooth use prior to the
update of `psmove pair`?

adangert, 2020-01-27:
So after some more testing after removing lines 777 - 780 the ZCM2 controllers will pair successfully on the pi. https://github.com/thp/psmoveapi/blob/master/src/platform/psmove_port_linux.cpp#L777

I've only been using the ZCM2 controllers with the raspberry pi previously, for some reason the controllers don't request a PIN code with the pi.

nitsch, 2020-01-27:
> I've only been using the ZCM2 controllers with the raspberry pi
> previously, for some reason the controllers don't request a PIN code
> with the pi.

This seems rather odd.

Could you fire up Wireshark for capturing the Bluetooth traffic between
controller and Pi, completely remove the previously paired controller
from the system (delete the stuff in /var/lib/bluetooth/<BT radio's
bdaddr>/ that belongs to the controller), start `psmove pair`?

There is either something running on the Pi that is answering the PIN
code request or, rather unlikely, the controller does indeed not request
a PIN code.

nitsch, 2020-01-27:
I will try to reproduce this as soon as I can get my hands on a
Raspberry Pi.

Which setup exactly are you using (hardware revision, operating system,
version of BlueZ, ...)?

adangert, 2020-01-27:
yeah I'm using a raspberry pi 4, with the Raspbian Buster with desktop option. This is the script I am using to install components and the psmoveapi: https://github.com/adangert/JoustMania/blob/master/setup.sh
other than that it's the base install of buster with no other modifications.

nitsch, 2020-02-02:
@adangert Looks like I will not have a Raspberry Pi 4 available anytime soon. So please take a look yourself using Wireshark. It might also be useful to test with a different Bluetooth radio.

adangert, 2020-02-16:
Ok, once I get some spare time, I will give it a look

LegendL3n, 2020-03-01:
I'm sorry if this is too off-topic but is it a Raspberry PI Zero W?
How did you compile the api in it? (having trouble doing so)

thp, 2020-03-01:
@LegendL3n Please create a separate issue for build instructions.

LegendL3n, 2020-03-01:
Ok, I've deleted the comment, sorry.

adangert, 2020-03-23:
@nitsch update on the problem: I've run wireshark on the pi to look at bluetooth packets, here is the output:
https://drive.google.com/open?id=1CKTxldSPqw9024MSz5LnjMqpwclXXMVB

messages 0 to 94 is when I type sudo ./psmove pair
the rest is after I press the playstation sync button on the controller.

adangert, 2020-03-24:
Found the issue was secure simple pairing, updated it in the pull request.

adangert, 2020-03-25:
Closing since #401 was merged

closed 2020-03-25 by adangert