thp/psmoveapi #404: issue pairing second ps4 move

Neefers, 2020-03-17:
Good Afternoon,

this error popped up while trying to pair.. my friend Aaron has been showing me the ropes, he guided me, thanks aaron.

pi@raspberrypi:~/psmoveapi/build $ sudo ./psmove pair
Connected controllers: 1
ioctl (GFEATURE): Broken pipe
[PSMOVE WARNING] No calibration file found (/etc/psmoveapi/00_00_b6_f0_20_00.calibration or /etc/psmoveapi/00_00_b6_f0_20_00.calibration)
ioctl (GFEATURE): Broken pipe
[PSMOVE WARNING] hid_get_feature_report failed, kernel issue? see https://github.com/thp/psmoveapi/issues/108
[PSMOVE CRITICAL] Assertion fail in _psmove_get_zcm2_calibration_blob: res == PSMOVE_CALIBRATION_SIZE
PSMove #1 connected via USB.
ioctl (GFEATURE): Broken pipe
Pairing of #1 failed.
Error reading/loading calibration data.

adangert, 2020-03-17:
So one of his controllers, pairs fine, the second one throws this error when trying to pair it.

nitsch, 2020-03-21:
Which version of psmoveapi are you using (version number or Git commit
hash)?

Which kernel are you running on (see the issue linked in the warning)?

Is the other controller (the ones that works) also a ZCM2?

nitsch, 2020-08-06:
@Neefers @adangert Any update on this? Is this still relevant?

adangert, 2020-08-06:
It might have been a bad controller, I don't have access to it, so I am not able to debug. However both PS3/PS4 controllers have been working on my end

closed 2020-10-12 by nitsch