thp/psmoveapi #405: Installation fails on arch linux

JacopoGobbi, 2020-04-25:
The installation fails on archlinux with the following error:
```
Package 'libusb', required by 'virtual:world', not found
```
My understanding is that the build fails due to the package `libusb` having a different name or installation path in Arch Linux.

This custom made scripts (AUR package) for Arch linux, are not working either:
https://github.com/asonix/psmoveapi-arch
This scripts also includes some patches that are outdated that I could update.

I am happy to contribute to this repo and try to test and to create a fix for those issues, but I may need a little help. 

Any suggestions on where I can to fix those issues?

thp, 2020-04-25:
Weird, it seems like `libusb` does exist in Arch Linux:

https://www.archlinux.org/packages/core/x86_64/libusb/

(I don't have any experience with Arch Linux)

nitsch, 2020-04-25:
I have absolutely no experience with Arch Linux either. What exactly did
you execute to install?

Posting the build log containing the error message would be helpful.

JacopoGobbi, 2020-04-25:
Arch Linux is not that particular, basically to install C++ applications works similar to other distro with maybe few extra things.
The AUR package is just a script that tells the system how to install the software. So rather than building from sources yourself you just download a script from a package repository and it will install the software for you following steps that are known to be working.

The steps I run are as following:
```
git clone --recursive https://github.com/thp/psmoveapi.git
cd psmoveapi/
mkdir build
cd build/
cmake ..
```

and the build log in the terminal is:
```
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.6.3") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Checking for module 'dbus-1'
--   Found dbus-1, version 1.12.16
-- Checking for module 'libudev'
--   Found libudev, version 245
-- Checking for module 'bluez>=5'
--   Found bluez, version 5.54

  Tracker
    Tracker library:  No (OpenCV not found)
    PS Eye support:   Yes
    Use PS3EYEDriver: No (disabled)
-- Checking for module 'libusb'
--   Package 'libusb', required by 'virtual:world', not found
CMake Error at /usr/share/cmake-3.17/Modules/FindPkgConfig.cmake:467 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.17/Modules/FindPkgConfig.cmake:647 (_pkg_check_modules_internal)
  src/utils/CMakeLists.txt:20 (pkg_check_modules)
  src/CMakeLists.txt:193 (include)
  CMakeLists.txt:17 (include)

-- Configuring incomplete, errors occurred!
See also "/home/<myuser>/psmoveapi/build/CMakeFiles/CMakeOutput.log".
See also "/home/<myuser>

/psmoveapi/build/CMakeFiles/CMakeError.log".
```

[CMakeError.log](https://github.com/thp/psmoveapi/files/4533537/CMakeError.log)
[CMakeOutput.log](https://github.com/thp/psmoveapi/files/4533538/CMakeOutput.log)

JacopoGobbi, 2020-04-25:
Also, these are my libusb-related installed packages:
```
$ pacman -Q | grep libusb
lib32-libusb 1.0.23-1
libusb 1.0.23-2
libusbmuxd 2.0.1-1
```
libusb is not split into normal and `-dev` like in Ubuntu, the standard package should contain both:
https://bbs.archlinux.org/viewtopic.php?pid=338240#p338240

yuripourre, 2020-04-25:
This is just a guess, could you please try adding a new line here?

https://github.com/thp/psmoveapi/blob/master/cmake/FindUSB1.cmake#L32
`usr/include` or `usr/include/libusb-1.0`

Based on:
https://www.archlinux.org/packages/core/x86_64/libusb/

nitsch, 2020-04-25:
So this is CMake failing to locate libusb. See the pkg_check_modules
call in src/utils/CMakeLists.txt. I suspect the problem is that you have
libusb-1.0 installed on your system while CMake is looking for libusb
(which would be version < 1.0).

Try changing that identifier to libusb-1.0 in the pkg_check_modules to
see if that fixes it.

nitsch, 2020-04-25:
I just realized that the code that actually needs libusb only works with
the old API, i.e. not libusb-1.0 but libusb. Since both APIs are
incompatible, simply linking against libusb-1.0 (after having made CMake
find it) won't work.

So you either need to install libusb or remove that part of the build.
It looks as if there already is an option for that (PSMOVE_USE_SIXPAIR),
but that won't stop CMake from demanding the library be installed.

nitsch, 2020-05-15:
@JacopoGobbi Did this solve your issue?

JacopoGobbi, 2020-05-18:
@nitsch Just tried it today on my laprop (before I was on my desktop) and by running cmake, it found `libusb 0.1.12` which I do not see installed. Maybe it is thanks to `libusb-compat 0.1.7-1`? I am surprised it built successfully.
I am also having other issues about installing but still trying to figure out (I know a bit about C but apparently not enough, especially to handle C++).
```
 ~/p/build   master   cmake ..
-- The C compiler identification is GNU 10.1.0
-- The CXX compiler identification is GNU 10.1.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.6.3") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Checking for module 'dbus-1'
--   Found dbus-1, version 1.12.16
-- Checking for module 'libudev'
--   Found libudev, version 245
-- Checking for module 'bluez>=5'
--   Found bluez, version 5.54

  Tracker
    Tracker library:  No (OpenCV not found)
    PS Eye support:   Yes
    Use PS3EYEDriver: No (disabled)
-- Checking for module 'libusb'
--   Found libusb, version 0.1.12

  Build configuration
    Debug build:      No
    Library license:  BSD (see README.md for details)

  Additional targets
    C example apps:   Yes
    OpenGL examples:  Yes
    C test programs:  Yes
    NavCon test:      Yes (using SDL2 joystick API)

  Language bindings
    Python:           No (SWIG not found)
    Java:             No (SWIG not found)
    C#:               No (SWIG not found)
    Processing:       No (SWIG not found)

-- Configuring done
-- Generating done
-- Build files have been written to: /home/<MY_USER>/psmoveapi/build
 ~/p/build   master   sudo pacman -Q | grep libusb
[sudo] password for <MY_USER>: 
lib32-libusb 1.0.23-1
libusb 1.0.23-2
libusb-compat 0.1.7-1
libusbmuxd 2.0.1-1
```

nitsch, 2020-05-18:
> it found `libusb 0.1.12` which I do not see installed. Maybe it is
> thanks to `libusb-compat 0.1.7-1`?

Yes, I would think so. Arch's libusb-compat points to
https://github.com/libusb/libusb-compat-0.1 which self-describes as:

"A compatibility layer allowing applications written for libusb-0.1 to
work with libusb-1.0. libusb-compat-0.1 attempts to look, feel, smell
and walk like libusb-0.1."

nitsch, 2020-08-06:
@JacopoGobbi Is this issue solved?

closed 2020-10-12 by nitsch