thp/psmoveapi #407: AIM Controller Support

peter-awesome, 2020-05-11:
Hi,

I really want to use the AIM controller for fps games.

From reading some comments other places it sounds like the packets are similar to v2 move controllers.

I'm going to have some down time soon if the basic work is parsing the packets. 

Let me know what the basic workflow for adding support is and I'll see what I can do after surgery.

yuripourre, 2020-05-12:
@peter-awesome I am not a maintainer or anything but I do remember following the work @thp's work in https://github.com/thp/psmoveapi/pull/372/files

I guess it has very useful information on how to accommodate a new controller.

nitsch, 2020-05-14:
> Let me know what the basic workflow for adding support is and I'll
> see what I can do after surgery.
There is no workflow for this. Just look around and make yourself
familiar with the code, ignoring camera- and tracker-related stuff.

Also, you could check out pull request #361 which added support for the
PS4 move controller. If the AIM controller is similiar, only minimal
additions should be necessary.

closed 2020-10-12 by nitsch