thp/psmoveapi #409: Cannot open move for retrieving serial - When psmove is started by a cronjob at reboot

LegendL3n, 2020-06-10:
Hi there (again),

I have a script that does a rainbow color thing to my PSMove and, for some reason, when I put it to run in a crontab at reboot, is does indeed run but when I press the pair button on my controller it gives me:

> [PSMOVE CRITICAL] Assertion fail in on_monitor_event: Cannot open move for retrieving serial!

However, if I kill the program and manually run the same script as the one cron is running, it works perfectly...
Also might be worth noting that, in both scenarios, if I connect the controller via USB it works as well.

I know this is way too subjective, and I'm sorry for that but this has been bugging me hard...

Thanks in advance!

LegendL3n, 2020-06-10:
If there's some more information you need, let me know and I'll provide.

thp, 2020-06-10:
Have you tried creating a systemd service that depends on (is started
after) Bluetooth and udev is up? It could be that by the time the cronjob
runs, the system isn't properly set up. A non-reliable dirty hack could
also be to just delay startup for a few seconds, but a systemd service with
the right dependencies should be faster, more efficient and more reliable.

thp, 2020-06-11:
Create a script that prints out:

```
ls -al /dev/hidraw*
hcitool dev
dmesg | grep hidraw
id
echo $PATH
ech $LD_LIBRARY_PATH
```

Then run this within the cronjob and in they way it works for you and post the results here (taking care to remove any private information that's not relevant for the bug report).

LegendL3n, 2020-06-11:
Weird, I changed my cron script to call that script and somehow it is now working... Removed the call to that and it's working still..

Well, thanks for you time anyways, and sorry for this weird behaviour.

closed 2020-06-11 by LegendL3n