thp/psmoveapi #412: How to install and setup?

gabrielcosi, 2020-08-10:
I'm new to python programming and I'd like to use but I can't figure out how to install it. I've already downloaded de lastest release  for windows x64. Also I can't install the PS3 Eye Camera, I'm using https://archive.org/details/CLEyeDriver5.3.0.0341Emuline but it only shows on the CL Eye Test and not in cv2. I know this is not related with the API but if someone can help me I'd really appreciate it. Thanks

nitsch, 2020-08-15:
There are a couple of Python examples in the directory examples/python/
within this repository. Even a small "Whack a Cube" game (ep_whack.py)
that Thomas put together for demonstrating the camera and orientation
tracking, input processing etc. These examples are not part of the
psmoveapi release archive though. Simply copy that directory from GitHub
if you do have not already have it on your computer.

Regarding the "installation":

Make sure the psmoveapi Python module is found by pointing the
environment variable PYTHONPATH to the sub-directory bindings/python/
that is part of the psmoveapi release.

Then make sure the actual psmoveapi library (used by the Python module)
is found by pointing the environment variable PSMOVEAPI_LIBRARY_PATH to
the sub-directory lib/ that is part of the psmoveapi release. Again, use
an absolute path.

Finally, connect your move controller via USB cable, open a Windows
console and run

  python <path to examples>/example_new_api.py

This should light up the controller's sphere and print a bunch of sensor
values that change when you move the controller. No camera yet. Try to
get that working first, then try the other examples.

There is a catch, though: There are currently 2 Python APIs (see
https://github.com/thp/psmoveapi/issues/338), and most of the examples
are for the old one. The old Python bindings are, however, not included
in the psmoveapi release, so you would need to build them yourself as
part of the library. Maybe somebody has already built and uploaded them
somewhere, I'd have to check.

@thp Any particular reason for not including the old SWIG-based Python
bindings in the release?

a-eye-tech, 2020-09-01:
@nitsch Thanks, that help me out. However, the sensor values I get all remain at 0:

<Vec3 (0.00, 0.00, 0.00)> -> <RGB (0.00, 0.50, 1.00)> usb: True bt: False
Update controller: <Controller #0 a0:ab:51:5b:cc:36> 2

The connected PS Move controller at least turns blue.

The sensor values also remain at zero even when I run example_new_api.exe from Visual Studio.

Can someone please help me with this issue?

nitsch, 2020-09-01:
@a-eye-tech Please open a new issue instead of hijacking this one here.

a-eye-tech, 2020-09-01:
@nitsch Ok, sorry about that.

closed 2020-10-12 by nitsch

heretique, 2021-01-09:
Hi guys, I have a similar issue (not being able to use the old python api) when using the latest release version 4.0.12 and Python 3.9.
I've installed the release to `c:\apps\psmoveapi\` and then I set the environment variable `PSMOVEAPI_LIBRARY_PATH` to point to the `lib` folder `c:\apps\psmoveapi\lib`. For my test I didn't altered the `PYTHONPATH` variable, instead I copied the python bindings directly in the same folder as my test .py file.
When I `import psmoveapi` all is well and good but when I `import psmove` as in some of the examples that come with the source code archive, I get some exceptions:
```
File "C:\apps\psmoveapi\bindings\python\psmove.py", line 14, in swig_import_helper
  return importlib.import_module(mname)
File "C:\Python39\Lib\importlib\__init__.py", line 127, in import_module
  return _bootstrap._gcd_import(name[level:], package, level)
File "C:\Python39\Lib\importlib\_bootstrap.py", line 1030, in _gcd_import
  return _find_and_load(name, _gcd_import)
File "C:\Python39\Lib\importlib\_bootstrap.py", line 1007, in _find_and_load
  return _find_and_load_unlocked(name, import_)
File "C:\Python39\Lib\importlib\_bootstrap.py", line 986, in _find_and_load_unlocked
  module = _load_unlocked(spec)
File "C:\Python39\Lib\importlib\_bootstrap.py", line 666, in _load_unlocked
  module = module_from_spec(spec)
File "C:\Python39\Lib\importlib\_bootstrap.py", line 565, in module_from_spec
  module = spec.loader.create_module(spec)
File "C:\Python39\Lib\importlib\_bootstrap_external.py", line 1108, in create_module
  module = _bootstrap._call_with_frames_removed(
File "C:\Python39\Lib\importlib\_bootstrap.py", line 228, in _call_with_frames_removed
  return f(*args, **kwds)

builtins.ImportError: DLL load failed while importing _psmove: The specified module could not be found.

During handling of the above exception, another exception occurred:

￼ File "e:\Workspace\PythonProjects\TestPsMove\src\test_dearpygui.py", line 7, in <module>
  import psmove
File "C:\apps\psmoveapi\bindings\python\psmove.py", line 17, in <module>
  _psmove = swig_import_helper()
File "C:\apps\psmoveapi\bindings\python\psmove.py", line 16, in swig_import_helper
  return importlib.import_module('_psmove')
File "C:\Python39\Lib\importlib\__init__.py", line 127, in import_module
  return _bootstrap._gcd_import(name[level:], package, level)

builtins.ImportError: DLL load failed while importing _psmove: The specified module could not be found.
```
I need to use the tracking feature to implement a 3D sculpting app and it seems that is available only with the old `psmove` api.
Any help would be appreciated. Thank you!

nitsch, 2021-01-12:
Please open a new ticket for your issue.

Which version of the psmoveapi release are you using, the x86 or the x64
build? Does your Python installation match that architecture?

heretique, 2021-01-13:
Created a new issue here #436.