thp/psmoveapi #414: How to use psmove pair utility?

a-eye-tech, 2020-08-31:
I was able to successfully build with Developer Command Prompt for VS 2017 abd 'call scripts/visualc/build_msvc.bat 2017 x86', but how do I use the 'psmove pair' utility? Sorry if this is a dumb question, this is new to me.

nitsch, 2020-09-01:
Please see the documentation:

  https://psmoveapi.readthedocs.io/en/latest/pairing.html

Also, the program tells you what to do.

BTW, are you sure you need the 32-bit build? That is what the 'x86' is
for. Use 'x64' if you are on a 64-bit system, using 64-bit libraries and
programs.

a-eye-tech, 2020-09-01:
@nitsch Thanks for the reply, but I'm still very confused. I know about the documentation, but I don't understand what exactly is the 'psmove pair' utility or how to use it. Is the 'psmove pair' utility something I would call in my code? Is it a .exe file? Is it something a call with Command Prompt? Something else?

I see psmove_pair(PSMove *move); is declared in psmove.h, is this the 'psmove pair' utility mentioned in the readthedocs?

The x86 is what I have been able to successfully install. Whenever I try to install the x64 version, I get this error:

C:\Users\user\source\repos\psmoveapi\external\opencv\modules\core\include\opencv2/core/hal/intrin_avx512.hpp(1769): err
or C2065: '_MM_PERM_ACBD': undeclared identifier [C:\Users\user\source\repos\psmoveapi\external\opencv\build-x64\module
s\features2d\opencv_features2d_AVX512_SKX.vcxproj]
Failed to build OpenCV (Debug)

nitsch, 2020-09-01:
`psmove` is a utility application (an .exe file in your case since you
seem to be using Windows) that is built together with the actual
library. You could implement it yourself using psmoveapi, but the
functionality is needed so often that this implementation is already
provided.

Check your build directory for psmove.exe.

a-eye-tech, 2020-09-01:
I ran psmove.exe from my \build-x86\, and the program window closes immediately. I ran psmove.exe from Visual Studio, and I get several 'Symbols Loaded' messages and 'psmove.exe' has exited with code 0 (0x0). Is this normal?

nitsch, 2020-09-01:
psmove.exe is a command-line application. Run it from a Command Prompt.

a-eye-tech, 2020-09-01:
Running "psmove.exe pair" on administrative Command Prompt did it. My controller now shows up as Motion Controller on my Bluetooth Devices! Thanks @nitsch !

closed 2020-09-05 by nitsch