thp/psmoveapi #415: ModuleNotFoundError: No module named 'psmove'

a-eye-tech, 2020-09-01:
This happens when I run always.py. I think I have SWIG installed correctly.

-  I have swigwin-4.0.2 in my C: drive
- I added C:\swigwin-4.0.2 to PATH
- restarted my computer
- I got _Build completed successfully_ when I ran call scripts/visualc/build_msvc.bat 2017 x86

Help please?

nitsch, 2020-09-05:
The error message means that Python cannot find the psmove.py file that
was generated as part of the library build. The file should be located
in your build output directory.

You can either move psmove.py and _psmove.pyd to a place that Python is
already searching for modules or you can add your build output directory
to the PYTHONPATH environment variable.

a-eye-tech, 2020-09-07:
@nitsch The files psmove.py and _psmove.pyd do not exist in my \build-x86\ directory. A search for those files in the folder \psmoveapi\ reveals nothing. The only files that look familiar are psmoveapi.py in \psmoveapi\bindings\python and psmove_test.py in psmoveapi\examples\python

Any ideas on how to proceed please?

nitsch, 2020-09-07:
Both files are generated as part of the build, so you won't find them in
the sources.

Check the build output to see whether building the Python bindings is
actually enabled. It is listed in the section "Language bindings" in
psmoveapi's CMake configure step. What does it say there?

a-eye-tech, 2020-09-08:
I think the file you speak of is CMakeLists.txt in \psmoveapi\bindings? This is one of the lines under " # Language Bindings":

`option(PSMOVE_BUILD_PYTHON_BINDINGS "Build the Python bindings" ON)`

nitsch, 2020-09-08:
> I think the file you speak of is CMakeLists.txt in \psmoveapi\bindings? This is one of the lines under " # Language Bindings":
> 
> `option(PSMOVE_BUILD_PYTHON_BINDINGS "Build the Python bindings" ON)`

No, I am talking about the console output that the build generates.

a-eye-tech, 2020-09-08:
Ok, I did "_call scripts/visualc/build_msvc.bat 2017 x86_"again and I found Language bindings. This is what I found:

```
  Language bindings
    Python:           No (libpython not found)
```

nitsch, 2020-09-08:
>   Language bindings
>     Python:           No (libpython not found)
Which tells you that no Python bindings were built because CMake could
not find libpython on your system.

Do you *have* Python installed?

a-eye-tech, 2020-09-08:
Yes, I have Python 3.6 installed, both 32-bit and 64-bit. I use Python often with my Visual Studio.

nitsch, 2020-09-08:
Is there any information on CMake's search for Python in the build output?

Which Git commit from the psmoveapi repository are you building?

Which version of CMake are you using?

Which version of Windows are you using?

a-eye-tech, 2020-09-09:
I'm not sure about CMake's search for Python in the build output, but here's all the lines that mention Python:
```
-- Could NOT find PythonInterp: Found unsuitable version "1.4", but required is at least "3.2" (found C:/Users/user/AppData/Local/Microsoft/WindowsApps/python3.exe)
-- OpenCV Python: during development append to PYTHONPATH: C:/Users/user/source/repos/psmoveapi/external/opencv/build-x86/python_loader
--     Unavailable:                 cudaarithm cudabgsegm cudacodec cudafeatures2d cudafilters cudaimgproc cudalegacy cudaobjdetect cudaoptflow cudastereo cudawarping cudev java js python2 python3 ts viz
--   Python (for build):            NO
Python:           No (libpython not found)
```

I used Visual Studio 2017 to make a clone of https://github.com/thp/psmoveapi , I think that answers the questions of which Git commit?

CMake 3.10.0

Windows 10 64-bit

nitsch, 2020-09-09:
> -- Could NOT find PythonInterp: Found unsuitable version "1.4", but required is at least "3.2" (found C:/Users/user/AppData/Local/Microsoft/WindowsApps/python3.exe)
> -- OpenCV Python: during [...]

That is all coming from the OpenCV build. Not what we are interested in.

You could try putting CMake in trace mode. In this line

https://github.com/thp/psmoveapi/blob/149696d4fb8a42697b9196b7060a13e3ee94d773/scripts/visualc/build_msvc.bat#L107

add the following options to the CMake call:

  --trace-expand --trace-source=FindPython.cmake

This should print lots of additional information regarding the search
for the Python libary when you run the script again.

> I used Visual Studio 2017 to make a clone of https://github.com/thp/psmoveapi , I think that answers the questions of which Git commit?

No, it depends on *when* you did that. Recent changes may be missing.

Make sure to fetch the latest commit which is currently 149696d.

nitsch, 2020-09-09:
> CMake 3.10.0

Oh, and the latest version of psmoveapi in the Git repository requires
at least CMake 3.12, so you should update.

a-eye-tech, 2020-09-10:
Ok, here's what I did:

* Updated CMake to 3.18.2
* Created a new clone on Visual Studio called psmoveapi2-10SEP2020. I created a new clone because I don't know how to fetch latest commit. Hopefully, creating a new clone as of 9/10/2020 has the latest commit?
* Added '--trace-expand --trace-source=FindPython.cmake' to line 107 of build_msvc.bat
* Successfully build x86 on psmoveapi2-10SEP2020

There's many mentions of Python on the x86 build output, many of them in the format <...>FindPython.cmake<...>

Even though the x86 build output has these lines:

```
  Language bindings
    Python:           Yes (version 3.6.4)
```
Running always.py still gives me the "No module named 'psmove'" error.

Was there a specific line you would like me to provide from my latest x86 build output?

nitsch, 2020-09-10:
> There's many mentions of Python on the x86 build output, many of them in the format <...>FindPython.cmake<...>

Those are the ones I was interested in. But we don't need them anymore for debugging because ...

> ```
>   Language bindings
>     Python:           Yes (version 3.6.4)
> ```

... the Python libraries were already found.

> Running psmoveapi2-10SEP2020\examples\python\always.py still gives me the "No module named 'psmove'" error.

Please make the changes I described in https://github.com/thp/psmoveapi/issues/415#issuecomment-687641831.

a-eye-tech, 2020-09-10:
Alright, things seem to be working just fine now for my Python and PS Move API. Thank you so much @nitsch !

closed 2020-09-10 by nitsch