thp/psmoveapi #416: sorry, unimplemented: non-trivial designated initializers not supported

BRCDel, 2020-09-03:
Building this on Manjaro. Cloned the repo using git clone, covered dependencies with Homebrew and pip2. Running the build_package.sh script (and by extension, debian_build in its own directory) throws the title's error in file psmoveapi/src/platform/psmove_port_linux.cpp - Line 308, column 1.

The bit of code that ends in line 308 is a const:

```cpp
const DBusObjectPathVTable pinagent_vtable = {
    .message_function = pinagent_message_handler
};
```

`pinagent_message_handler` is also mentioned in the above block (Lines 293 to 304)

```cpp
pinagent_message_handler(DBusConnection *conn, DBusMessage *msg, void *data)
{
    if (dbus_message_is_method_call(msg, "org.bluez.Agent1", "RequestPinCode")) {
        return pinagent_request_pincode_message(conn, msg, data);
    }

    if (dbus_message_is_method_call(msg, "org.bluez.Agent1", "Release")) {
        return pinagent_release_message(conn, msg, data);
    }

    return DBUS_HANDLER_RESULT_NOT_YET_HANDLED;
}```

GCC 8 is claimed to support non-trivial designated initializers but downgrading and installing from scratch did not solve the issue.

Any clue how I should get past this?

BRCDel, 2020-09-03:
note: Homebrew apparently uses gcc5. I'll look into that possibly being the source of the issue, but I'm at a loss right now.

BRCDel, 2020-09-03:
Second note: It's Brew, and it doesn't want to use gcc@8. This is all pointing to being an external issue.

BRCDel, 2020-09-03:
I made it work. The (admittedly janky as hell) solution was to delete c++, g++ and gcc in /home/linuxbrew/.linuxbrew/bin and rename c++-8, g++-8 and gcc-8 to c++, g++ and gcc respectively.

closed 2020-09-03 by BRCDel

nitsch, 2020-09-04:
Which compiler was that, initially? The exact version, please.

The problem is: this is C code that was later added to this file which
is treated as C++ (because it uses C++ in the other parts). The
initialization of pinagent_vtable is valid C99 but not valid C++. At
least it *was*, because C++ recently added a similar thing which would
make this particular case work.

I never realized the mistake because gcc 8 already supported the new C++
feature. You, on the other hand, seem to be using an older version.

BRCDel, 2020-09-04:
> ​
>
> CAUTION: This email originated from outside Edinburgh Napier University. Do not follow links or open attachments if you doubt the authenticity of the sender or the content.

Hello,

Brew installed gcc 5.5.0 (gcc@5) on my system instead of the default gcc 10.2.0 (gcc@10). This error completely lies with either me or Brew and not you - Admittedly, even about an hour into troubleshooting, submitting an error report was a rushed move. This was just a mess up on my end. Thank you so much for creating this API. Consider this a non-issue.

Kind regards,
Del.

This message and its attachment(s) are intended for the addressee(s) only and should not be read, copied, disclosed, forwarded or relied upon by any person other than the intended addressee(s) without the permission of the sender. If you are not the intended addressee you must not take any action based on this message and its attachment(s) nor must you copy or show them to anyone. Please respond to the sender and ensure that this message and its attachment(s) are deleted.

It is your responsibility to ensure that this message and its attachment(s) are scanned for viruses or other defects. Edinburgh Napier University does not accept liability for any loss or damage which may result from this message or its attachment(s), or for errors or omissions arising after it was sent. Email is not a secure medium. Emails entering Edinburgh Napier University's system are subject to routine monitoring and filtering by Edinburgh Napier University.

Edinburgh Napier University is a registered Scottish charity. Registration number SC018373

BSL users can contact us via contactSCOTLAND-BSL, the on-line British Sign Language interpreting service. Find out more on the contactSCOTLAND website.