thp/psmoveapi #417: No module named 'psmoveapi'

a-eye-tech, 2020-09-16:
I am able to run '_example_new_api.py_' from my Visual Studio 2017 clone. I can also run '_example_new_api.py_' with the Command Prompt command:

'_C:\Users\user\source\repos\psmoveapi2-10SEP2020\examples\python example_new_api.py_'

I have the line '_C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86_' in my PYTHONPATH.

However, when I copy and the paste all the text in 'example_new_api.py' into a new Visual Studio 2017 project, I get the error '_No module named 'psmoveapi'_'

Any ideas? One thing I noticed is that all the Visual Studio projects with 'example_new_api.py' have the a warning on 'import psmoveapi'. The warning being, "_Unable to resolve 'psmoveapi'. Intelligence may be missing from this module_'.

a-eye-tech, 2020-09-16:
Update:

I found the file 'psmoveapi.py' in the directory '<...>/bindings/python'. I made a copy of 'psmoveapi.py' into my new Visual Studio 2017 Python project's folder, and the code seems to be running just fine. 

I still get the _"Unable to resolve 'psmoveapi'. Intelligence may be missing from this module."_ warning.

nitsch, 2020-09-17:
> I have the line '_C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86_' in my PYTHONPATH.

As you already found out, you need to point Python to the psmoveapi
Python bindings that are found in bindings/python/psmoveapi.py, not in
the build directory.

These bindings then use the psmoveapi library (which is built in the
build directory). Have a look at psmoveapi.py and example_new_api.py and
you will find that they check an environment variable named
PSMOVEAPI_LIBRARY_PATH in order to locate the psmoveapi library.

> I still get the _"Unable to resolve 'psmoveapi'. Intelligence may be 
> missing from this module."_ warning.

I have absolutely zero experience with Python in Visual Studio. But I am
guessing that you somehow have to add a reference to psmoveapi.py to
your project. As dependency, library or whatever Visual Studio may call it.

a-eye-tech, 2020-09-22:
I don't know what I'm doing wrong. Here's what my environment variables looks like:

```
PSMOVEAPI_LIBRARY_PATH:
C:\Users\user\source\repos\psmoveapi2-10SEP2020\bindings\python
C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86
```

```
PYTHONPATH:
C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86
C:\Users\user\source\repos\psmoveapi2-10SEP2020\bindings\python
```

I started a new Python project that looks like this:

```
import os
import sys
import time

BASE = os.path.join(os.path.dirname(__file__), '..', '..')

if 'PSMOVEAPI_LIBRARY_PATH' not in os.environ:
    os.environ['PSMOVEAPI_LIBRARY_PATH'] = os.path.join(BASE, 'build')

sys.path.insert(0, os.path.join(BASE, 'bindings', 'python'))

import psmoveapi
import psmove

move = psmove.psmove_connect()
```

But I still get the error _"No module named 'psmoveapi'"._

Any thoughts?

nitsch, 2020-09-22:
> ```
> PSMOVEAPI_LIBRARY_PATH:
> C:\Users\user\source\repos\psmoveapi2-10SEP2020\bindings\python
> C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86
> ```

This must point to the directory that contains the psmoveapi.dll. Why are there two directories listed?

> ```
> PYTHONPATH:
> C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86
> C:\Users\user\source\repos\psmoveapi2-10SEP2020\bindings\python
> ```

This must point to the directory that contains [psmoveapi.py](https://github.com/thp/psmoveapi/blob/4ab0456cf284bac5de7d028472b7009cd7f79a51/bindings/python/psmoveapi.py). Why are there two directories listed?

> ```
> import psmoveapi
> import psmove
> ```

You should not mix the Python bindings. Either use psmove or psmoveapi, not both of them at the same time.

> Any thoughts?

The path to the Python bindings is likely wrong. That's why the module cannot be loaded.

For testing, don't use any Python projects in Visual Studio or any other IDE, just create your Python script in a text editor and run it in Python from the command line. (Alternatively, if you absolutely must use an IDE, print the contents of both environment variables in your Python script and verify that they are pointing to the correct locations.)

Use [example_new_api.py](https://github.com/thp/psmoveapi/blob/4ab0456cf284bac5de7d028472b7009cd7f79a51/examples/python/example_new_api.py) as is, *without any modifications* until you got that working.

thp, 2020-09-23:
1. Set `PSMOVEAPI_LIBRARY_PATH` pointing to where your DLL is
2. Set `PYTHONPATH` pointing to where your `psmoveapi.py` is
3. Read the documentation and examples

closed 2020-09-23 by thp

a-eye-tech, 2020-09-27:
I'm sorry, but I'm still struggling with this problem even though I believe I'm appropriately reading the documentation and examples. 

I changed my Environment Variables so they look like this:

```
PSMOVEAPI_LIBRARY_PATH:
C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86
```
This directory contains the file _psmoveapi.dll_

```
PYTHONPATH:
C:\Users\user\source\repos\psmoveapi2-10SEP2020\bindings\python
```
This directory contains the file _psmoveapi.py_

I added these lines to the code:
```
print("PSMOVEAPI_LIBRARY_PATH: ", os.environ['PSMOVEAPI_LIBRARY_PATH'])
print("PYTHONPATH: ",os.environ['PYTHONPATH'])
```

_print("PSMOVEAPI_LIBRARY_PATH: ", os.environ['PSMOVEAPI_LIBRARY_PATH'])_
Printed path matches PSMOVEAPI_LIBRARY_PATH in Windows 10 Environment Variables

_print("PYTHONPATH: ",os.environ['PYTHONPATH'])_
Printed path matches PSMOVEAPI_LIBRARY_PATH in Windows 10 Environment Variables

However, when I run PS_Move_Tests.py on Visual Studio 2017:
_print("PYTHONPATH: ",os.environ['PYTHONPATH'])_
Printed path DOES NOT match PYTHONPATH in Windows 10 Environment Varaibles.

Also, for some reason I am unable to run example_new_api.py, always.py, or any of the other python files from my PS Move clone either from Command Prompt of Visual Studio.

nitsch, 2020-09-27:
You have all the Python sources and the error message right there, with
line numbers. Just go in there and find out which DLL ctypes is trying
to load (and failing). If that filename and path does not match the
location of your psmoveapi.dll, then your PSMOVEAPI_LIBRARY_PATH is most
likely wrong.

a-eye-tech, 2020-10-02:
I modified the files _example_new_api.py_, _psmoveapi.py_, and ___init__.py_ so that it prints out the variables in the line number where I get the errors. I still don't know what the problem is. I verified the that files are where they are suppose to be according the PYTHONPATH and PSMOVEAPI_LIBRARY_PATH.

The only thing that looks kinda off to me is:

library_prefix+'psmoveapi.dll' =  C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86;\psmoveapi.dll

But other than that, I have no idea what to do even after many hours of experimenting and trying to figure out what's wrong.

```
C:\Users\user\source\repos\psmoveapi2-10SEP2020\examples\python>python example_new_api.py

example_new_api.py:
PSMOVEAPI_LIBRARY_PATH:  C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86;
PYTHONPATH:  C:\Users\user\source\repos\psmoveapi2-10SEP2020\bindings\python;

__init__.py:
self._name =  python dll
mode =  None

__init__.py:
self._name =  kernel32
mode =  0

psmoveapi.py:
PSMOVEAPI_LIBRARY_PATH:  C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86;
library_prefix =  C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86;\
library_prefix+'psmoveapi.dll' =  C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86;\psmoveapi.dll

__init__.py:
self._name =  C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86;\psmoveapi.dll
mode =  0

Traceback (most recent call last):
  File "example_new_api.py", line 18, in <module>
    import psmoveapi
  File "..\..\bindings\python\psmoveapi.py", line 48, in <module>
    libpsmoveapi = CDLL(library_prefix + 'psmoveapi.dll')
  File "C:\Python36-32\lib\ctypes\__init__.py", line 352, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found
```

nitsch, 2020-10-02:
> psmoveapi.py:
> PSMOVEAPI_LIBRARY_PATH:  C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86;
> library_prefix =  C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86;\
> library_prefix+'psmoveapi.dll' =  C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x86;\psmoveapi.dll

You have the answer right there. The trailing semicolon in
PSMOVEAPI_LIBRARY_PATH is screwing up the path to the psmoveapi.dll. Not
sure why you put it in there in the first place, but you should
definitely remove it.

a-eye-tech, 2020-10-02:
Ok I noticed that on Environment Variables, there is a semicolon placed automatically at the end of my PSMOVEAPI_LIBRARY_PATH and PYTHONPATH. When I edit these lines, the semicolons are not shown.

I solved this issue by deleting PSMOVEAPI_LIBRARY_PATH and PYTHONPATH, and creating them again. This time, these variables did not have the semicolon shown in Environment Variables. 

Thanks @nitsch !

Hopefully, I can start creating PS Move Python projects.