thp/psmoveapi #418: 'Undeclared Identifier' when on my x64 build attempt.

a-eye-tech, 2020-09-17:
When I run this command, 'call scripts/visualc/build_msvc.bat 2017 x64', I get the following error text in red:

```
C:\Users\user\source\repos\psmoveapi2-10SEP2020\external\opencv\modules\core\include\opencv2/core/hal/intrin_avx512.hpp
(1769): error C2065: '_MM_PERM_ACBD': undeclared identifier [C:\Users\user\source\repos\psmoveapi2-10SEP2020\external\o
pencv\build-x64\modules\core\opencv_core_AVX512_SKX.vcxproj]
C:\Users\user\source\repos\psmoveapi2-10SEP2020\external\opencv\modules\core\include\opencv2/core/hal/intrin_avx512.hpp
(1769): error C2065: '_MM_PERM_ACBD': undeclared identifier [C:\Users\user\source\repos\psmoveapi2-10SEP2020\external\o
pencv\build-x64\modules\features2d\opencv_features2d_AVX512_SKX.vcxproj]
C:\Users\user\source\repos\psmoveapi2-10SEP2020\external\opencv\modules\core\include\opencv2/core/hal/intrin_avx512.hpp
(1769): error C2065: '_MM_PERM_ACBD': undeclared identifier [C:\Users\user\source\repos\psmoveapi2-10SEP2020\external\o
pencv\build-x64\modules\dnn\opencv_dnn_AVX512_SKX.vcxproj]
C:\Users\user\source\repos\psmoveapi2-10SEP2020\external\opencv\modules\core\include\opencv2/core/hal/intrin_avx512.hpp
(1769): error C2065: '_MM_PERM_ACBD': undeclared identifier [C:\Users\user\source\repos\psmoveapi2-10SEP2020\external\o
pencv\build-x64\modules\imgproc\opencv_imgproc_AVX512_SKX.vcxproj]
```
Followed eventually by:

`Failed to build OpenCV (Debug)`

Could this error have to do that I have already have a few OpenCV 64-Bit builds that I use with my OpenCV Visual Studio 2017 C++ projects?

I am able to successfully build and run the PS Move API x86 build, but I would really like to be able to use the PS Move API x64 build.

nitsch, 2020-09-17:
> Could this error have to do that I have already have a few OpenCV 64-Bit builds that I use with my OpenCV Visual Studio 2017 C++ projects?

I doubt it. This is solely an OpenCV build issue. Other people seem to
be having the same problem (see
https://github.com/microsoft/vcpkg/issues/13122 for example). Please
check which version of OpenCV you are trying to build (see the build
log) and consult their issue tracker for help.

But then again, you most likely do not even have to build OpenCV
yourself. If you already have OpenCV libraries installed on your system,
use those instead. In scripts/visualc/build_msvc.bat, throw out the
parts that clone and build OpenCV, then point psmoveapi's initial
configuration step
https://github.com/thp/psmoveapi/blob/149696d4fb8a42697b9196b7060a13e3ee94d773/scripts/visualc/build_msvc.bat#L107
to the location of your OpenCV installation.

a-eye-tech, 2020-09-17:
I changed line 107 in _build_msvc.bat_ so it looks like this:
`cmake .. -G "%MSVC_CMAKE_GENERATOR%" -DPSMOVE_USE_PS3EYE_DRIVER=1 -DPSMOVE_BUILD_OPENGL_EXAMPLES=1 -DOpenCV_DIR=C:/OpenCV3-4-2/opencv/build/ --trace-expand --trace-source=FindPython.cmake`

Then I deleted these lines in _build_msvc.bat_:
```
REM Clone OpenCV
IF NOT EXIST %OPENCV_DIR% (
	cd %PSMOVE_API_EXTERNAL_DIR%
	git clone --depth 1 --branch 3.4 git://github.com/opencv/opencv.git
) ELSE (
	echo.
	echo OpenCV dir already exists; assuming it has been cloned already
)
```

Then I ran the code:
`call scripts/visualc/build_msvc.bat 2017 x64`

During the build I got this error in red:
```
LINK : fatal error LNK1104: cannot open file 'python36_d.lib' [C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x6
4\_psmove.vcxproj]
```
Followed eventually by:
`Failed to build PSMoveAPI (Debug)`

However, the directory _<...>psmoveapi2-10SEP2020\build-x64_ was created with the file _psmoveapi.dll_. I copied this _psmoveapi.dll_ file from _\build-x64_ to my Visual Studio 2017 C++ project titled _x64test_. The of the copied _psmoveapi.dll_ is in the same location as my x64text.exe. I ran my x64test Visual Studio project and it seems to be running fine.

Thanks for you help @nitsch ! 

Hopefully, the error I got during build is not too serious.

nitsch, 2020-09-18:
> Then I deleted these lines in _build_msvc.bat_:

If you really want to stop building OpenCV in this script, you should also delete the part that actually builds OpenCV. So far you have only stopped the script from cloning the OpenCV Git repository.

> ```
> LINK : fatal error LNK1104: cannot open file 'python36_d.lib' [C:\Users\user\source\repos\psmoveapi2-10SEP2020\build-x6
> 4\_psmove.vcxproj]
> ```

This looks like an error during the debug build of the old SWIG-based Python bindings. If you don't need them, throw them out (add `-DPSMOVE_BUILD_PYTHON_BINDINGS=0` to [psmoveapi's configure step](https://github.com/thp/psmoveapi/blob/149696d4fb8a42697b9196b7060a13e3ee94d773/scripts/visualc/build_msvc.bat#L107).). If you want to build them, make sure there actually is a python36_d.lib installed on your system. You could open the generated _psmove.vcxproj in Visual Studio and check out where it is looking for that library.

a-eye-tech, 2020-09-22:
To stop building OpenCV in build_msvc.bat, would I need to remove these lines?:

```
REM Generate OpenCV solution

echo.
echo Generating OpenCV solution

cd %OPENCV_DIR%
IF NOT EXIST build-%MSVC_PLATFORM% mkdir build-%MSVC_PLATFORM%
cd build-%MSVC_PLATFORM%

cmake .. -G "%MSVC_CMAKE_GENERATOR%" -DBUILD_SHARED_LIBS=0 -DBUILD_WITH_STATIC_CRT=OFF -DBUILD_PERF_TESTS=OFF -DBUILD_TESTS=OFF -DBUILD_DOCS=OFF -DBUILD_opencv_apps=OFF -DBUILD_opencv_flann=ON -DBUILD_opencv_features2d=ON -DBUILD_opencv_objdetect=OFF -DBUILD_opencv_photo=OFF -DBUILD_opencv_ts=OFF -DBUILD_opencv_ml=ON -DBUILD_opencv_video=OFF -DBUILD_opencv_java=OFF -DWITH_OPENEXR=OFF -DWITH_FFMPEG=OFF -DWITH_JASPER=OFF -DWITH_TIFF=OFF
IF !ERRORLEVEL! NEQ 0 (
	echo Failed to generate OpenCV solution
	goto Error
)

REM Build OpenCV
echo.
echo Building OpenCV ^(Debug^)
msbuild.exe %OPENCV_DIR%/build-%MSVC_PLATFORM%/ALL_BUILD.vcxproj /p:Configuration=Debug /property:Platform=%MSVC_PLATFORM% /verbosity:minimal /maxcpucount
IF %ERRORLEVEL% NEQ 0 (
	echo Failed to build OpenCV ^(Debug^)
	goto Error
)
echo.
echo Building OpenCV ^(Release^)
msbuild.exe %OPENCV_DIR%/build-%MSVC_PLATFORM%/ALL_BUILD.vcxproj /p:Configuration=Release /property:Platform=%MSVC_PLATFORM% /verbosity:minimal /maxcpucount
IF %ERRORLEVEL% NEQ 0 (
	echo Failed to build OpenCV ^(Release^)
	goto Error
)
```

I found '_psmove.vcxproj' in <...>\build-x64. I opened '_psmove.vcxproj' and I opened the properties on '_psmove'. This is what is listed in 'Linker-Additional Dependencies':

`C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64\libs\python36.lib;<different options>`

I verified that 'pthon36.lib' does indeed exist in that directory.

nitsch, 2020-09-22:
> To stop building OpenCV in build_msvc.bat, would I need to remove these lines?: [...]

Yes.

> However, my build error mentions 'python36_d.lib', which exists in
> my location of 'C:\Python36-32\libs'
Then you should try to find out where in that project the reference to
python36_d.lib is coming from. Or you just skip the debug build of
psmoveapi for now if you do not need it.

Also, you said that this was an x64 build. Linking the x86 version of
the Python debug library won't work.

a-eye-tech, 2020-09-22:
How do I stop the debug build? Do I need to make a change in _'build_msvc.bat'_?

nitsch, 2020-09-22:
> How do I stop the debug build? Do I need to make a change in _'build_msvc.bat'_?

Yes, in there.

closed 2020-10-12 by nitsch