thp/psmoveapi #425: test_opengl.cpp gives error "Assertion fail in ps_move_tracker_get_size: tracker != NULL"

a-eye-tech, 2020-09-25:
When I try to run _test_opengl.cpp_, I get the following error:

```
[PSMOVE CRITICAL] Assertion fail in psmove_tracker_get_size: tracker != NULL
Assertion failed: width > static_cast<T>(0), file c:\users\user\source\repos\psmoveapi2-10sep2020\external\glm\glm\gtc\matrix_transform.inl, line 316
```

Can someone please help me with this issue? The same thing happens when I run _test_opengl2.cpp_ and _test_opengl3.cpp_. I am using Visual Studio 2017 on Windows 10.

nitsch, 2020-09-26:
Looks like the OpenGL examples are doing poor error handling.

You probably do not have a camera connected/enabled.

a-eye-tech, 2020-09-26:
Ok I have my PS3 camera connect and I also installed CL PS3 Camera Driver. Looks like I made progress, but I get this warning:

`Warning: No lens calibration files found.`

and nothing happens, except for my move controller bulb intermittently turning pink.

nitsch, 2020-09-26:
The tracker is probable unsuccessfully trying to locate the glowing orb.
Put the controller in front of the camera so that it is in the frame.

a-eye-tech, 2020-09-26:
I tried putting the controller in front of the camera at various distances, but still makes no difference.

nitsch, 2020-09-26:
Make sure this is the actual camera psmoveapi is using. If it is a
laptop, maybe you are actually using the built-in camera. See the
documentation: https://psmoveapi.readthedocs.io/en/latest/config.html

Also, try the simpler test_tracker example first.

a-eye-tech, 2020-09-26:
Ok, I removed all the cameras attached to my PC except for the PS3 camera. Everything seems to be working properly, except the OpenGL examples seem to be slow and jittery (maybe this can be addressed by modifying PSMOVE_TRACKER_ROI_SIZE?).

Thanks for the help @nitsch !

closed 2020-10-02 by nitsch