thp/psmoveapi #427: Can't add the C# DLL to Visual Studio

LegendL3n, 2020-11-04:
Hi there.

I'm trying to use psmove in a C# project, however I get this error when adding the DLL:
[image]

Also don't I need another file? Or am I supposed to include the .lib?

Thanks in advance.

nitsch, 2020-11-07:
Adding psmoveapi_csharp.dll as reference is the way it is supposed to
work. The assembly also requires the psmoveapi.dll to be found.

If that does not work, there may a problem with the C# bindings. Which
version of psmoveapi are you using? And which version of Windows and
Visual Studio?

LegendL3n, 2020-11-07:
That's the DLL I'm trying to reference.
The version of psmoveapi is the [latest release](https://github.com/thp/psmoveapi/releases/tag/4.0.11).
Windows 10 and Visual Studio 2019 16.7.7

nitsch, 2020-11-18:
> Adding psmoveapi_csharp.dll as reference is the way it is supposed to work.

I stand corrected. I looks as if the psmoveapi build is spitting out a bunch of .cs files that are supposed to be used in your C# project instead. One of them loads psmoveapi_csharp.dll which is not even a .NET assembly.

However, these .cs files are currently missing from the release package.

nitsch, 2020-11-27:
The missing files have been added. There is no official psmoveapi release yet, but you could download the archive from the latest build's artifacts on the [CI server](https://ci.appveyor.com/project/thp/psmoveapi).

closed 2020-11-27 by nitsch