thp/psmoveapi #433: cmake build error on linux ubuntu20.04 about v4l2

mhw-Parker, 2020-12-15:
recently i met a problem when i want to use v4l2 with opencv on ubuntu20.04
here is the problem:
`/usr/bin/ld: CMakeFiles/test0.dir/rmvideocapture.cpp.o: in function `RMVideoCapture::RMVideoCapture(int, int)':
rmvideocapture.cpp:(.text+0x474): undefined reference to `v4l2_open'
/usr/bin/ld: CMakeFiles/test0.dir/rmvideocapture.cpp.o: in function `RMVideoCapture::cOpen(int, int)':
rmvideocapture.cpp:(.text+0xaa9): undefined reference to `v4l2_open'
/usr/bin/ld: CMakeFiles/test0.dir/rmvideocapture.cpp.o: in function `RMVideoCapture::setExposureTime(int)':
rmvideocapture.cpp:(.text+0x1350): undefined reference to `v4l2_set_control'
/usr/bin/ld: CMakeFiles/test0.dir/rmvideocapture.cpp.o: in function `RMVideoCapture::changeVideoFormat(int, int, bool)':
rmvideocapture.cpp:(.text+0x1726): undefined reference to `v4l2_close'
/usr/bin/ld: rmvideocapture.cpp:(.text+0x1736): undefined reference to `v4l2_open'
/usr/bin/ld: CMakeFiles/test0.dir/rmvideocapture.cpp.o: in function `RMVideoCapture::restartCapture()':
rmvideocapture.cpp:(.text+0x1a1c): undefined reference to `v4l2_close'
/usr/bin/ld: rmvideocapture.cpp:(.text+0x1a2c): undefined reference to `v4l2_open'
collect2: error: ld returned 1 exit status`
i have checked the header and the v4l2_open could be found in `#include <libv4l2.h>` 
and the clion can detect v4l2_open
whatever i do it just build failed
i would appreciate that if you can solve this problem, thx!

thp, 2020-12-15:
Did you install `libv4l-dev`?

thp, 2020-12-19:
@mhw-Parker Any updates on this?

mhw-Parker, 2021-03-30:
yes i have installed, i solved it later. thx

closed 2021-04-27 by nitsch