thp/psmoveapi #436: Couldn't use Python bindings on Windows with build 4.0.12 (64 bit)

heretique, 2021-01-13:
Hi guys, I'm not being able to use the old python api `import psmove` when using the latest release version 4.0.12 (64 bit) and Python 3.9, Python 3.8, Python 2.7 (all 64 bit) on Windows 10.

I've installed the release to `c:\apps\psmoveapi\` and then I set the environment variable `PSMOVEAPI_LIBRARY_PATH` to point to the lib folder `c:\apps\psmoveapi\lib`. For my test I didn't altered the `PYTHONPATH` variable, instead I copied the python bindings directly in the same folder as my test .py file.
When I `import psmoveapi` all is well and good but when I `import psmove` in my test file, I get some exceptions:
```
File "e:\Workspace\PythonProjects\TestPsMove\src\psmove.py", line 14, in swig_import_helper
  return importlib.import_module(mname)
File "C:\Python39\Lib\importlib\__init__.py", line 127, in import_module
  return _bootstrap._gcd_import(name[level:], package, level)
File "C:\Python39\Lib\importlib\_bootstrap.py", line 1030, in _gcd_import
  return _find_and_load(name, _gcd_import)
File "C:\Python39\Lib\importlib\_bootstrap.py", line 1007, in _find_and_load
  return _find_and_load_unlocked(name, import_)
File "C:\Python39\Lib\importlib\_bootstrap.py", line 986, in _find_and_load_unlocked
  module = _load_unlocked(spec)
File "C:\Python39\Lib\importlib\_bootstrap.py", line 666, in _load_unlocked
  module = module_from_spec(spec)
File "C:\Python39\Lib\importlib\_bootstrap.py", line 565, in module_from_spec
  module = spec.loader.create_module(spec)
File "C:\Python39\Lib\importlib\_bootstrap_external.py", line 1108, in create_module
  module = _bootstrap._call_with_frames_removed(
File "C:\Python39\Lib\importlib\_bootstrap.py", line 228, in _call_with_frames_removed
  return f(*args, **kwds)

builtins.ImportError: DLL load failed while importing _psmove: The specified module could not be found.

During handling of the above exception, another exception occurred:

File "e:\Workspace\PythonProjects\TestPsMove\src\test_dearpygui.py", line 7, in <module>
  import psmove
File "e:\Workspace\PythonProjects\TestPsMove\src\psmove.py", line 17, in <module>
  _psmove = swig_import_helper()
File "e:\Workspace\PythonProjects\TestPsMove\src\psmove.py", line 16, in swig_import_helper
  return importlib.import_module('_psmove')
File "C:\Python39\Lib\importlib\__init__.py", line 127, in import_module
  return _bootstrap._gcd_import(name[level:], package, level)

builtins.ImportError: DLL load failed while importing _psmove: The specified module could not be found.
```
I need to use the tracking feature to implement a 3D sculpting app and it seems that is available only with the old psmove api.
Any help would be appreciated. Thank you!

_PS: I managed to use the API from C++ but the tracker doesn't connect to my web camera (I only managed to connect the camera only when using an OpenCV4 based app, couldn't manage to do it with OpenCV3 but that's a different issue than what I described above)_

nitsch, 2021-01-13:
From your error message it seems that psmove.py fails to load the _psmove.pyd module file. You said you copied the Python bindings over to your folder. Make sure there is the _psmove.pyd in that folder.

You can also try to load the module manually. Open a Python promp in that directory and execute
```python
import importlib
importlib.import_module('_psmove')
```
Does that work? If it does, it means that psmove.py was looking for the module in the wrong place previously and you need to fix the search path (by appending the directory location to Python's `sys.path` for example).

heretique, 2021-01-14:
I've copied the `.pyd` file as well along with the others. Looking at the stack after the exception it is clear that the `.pyd` file is properly found.
This is how my test code looks:
```
import os
import sys
import pathlib

#sys.path.insert(0, os.path.join(os.environ['PSMOVEAPI_LIBRARY_PATH'], '..','bindings', 'python'))
curPath = pathlib.Path().absolute()
sys.path.insert(0, curPath)

import importlib
importlib.import_module('_psmove')
#import psmove
```
and this is the last call that fails:
[image]
I underlined the path, which is exactly the path of the `.pyd` file. That's the reason I'm so confused.

cboulay, 2021-01-14:
That error is just as often about not finding a dependency as it is about not finding the module itself.

Try downloading and running the latest [DependenciesGUI.exe](https://github.com/lucasg/Dependencies/releases), opening the .pyd file, and see what's missing.
There's a trick to getting accurate results from Dependencies when testing Python modules: you need to run DependenciesGUI in the same environment (especially the PATH) that your Python process would run in. So if you use Anaconda, then you need to run Dependencies from an Anaconda prompt with the correct environment activated.

thp, 2021-01-14:
You likely also need the PS Move API library (`libpsmoveapi.dll`) and the Tracker library (`libpsmoveapi_tracker.dll`). Just putting them in the same folder as `_psmove.pyd` should make them available. Also keep in mind that you need to have matching CPU architecture on the libraries, which means if Python is 32-bit and `_psmove.pyd` is 32-bit, then `libpsmoveapi.dll` and `libpsmoveapi_tracker.dll` also need to be 32-bit (and same if Python is 64-bit, then `_psmove.pyd` needs to be 64-bit and `libpsmoveapi.dll` and `libpsmoveapi_tracker.dll`, too). Also, make sure that `_psmove.pyd` has been compiled for/with the same Python version that you are using, otherwise there could be some mismatch or crash.

heretique, 2021-01-14:
I already have everything in the same folder
[image]
and I downloaded and installed x64 variant for everything

[image]
[image]

I don't know what Python version was used to build the Windows release that is available to download on Github, it is not specified. I also don't know exactly how to tell if the binaries in the psmoveapi release are actually x64 and not x86, I downloaded the x64 one.

heretique, 2021-01-14:
Thanks for all the help guys. I managed to get it working after @cboulay's suggestion to use the dependency viewer. 

[image]

I've noticed that it needs a `python38.dll` and I've tried again with `Python 3.8` and now it is being imported properly. I did test with `Python 3.8` before but probably in the confusion I didn't had all the dependencies in the same place and I figured I should continue testing with `Python 3.9` if it doesn't work with `Python 3.8`, since that's the one on my path.

closed 2021-01-14 by heretique