thp/psmoveapi #437: Python binding for release 4.0.12 doesn't include tracker support?

heretique, 2021-01-14:
Hi guys, after struggling to get the latest release working for me (my fault, #436) I've noticed that I couldn't use the tracker module with Python. Is this intentional? I suppose that those bindings should be available since there are Python examples using them.
This is what I get when trying to run the `test_tracker.py`:
```
File "e:\Workspace\PythonProjects\TestPsMove\src\test_tracker.py", line 41, in <module>
  tracker = psmove.PSMoveTracker()

builtins.AttributeError: module 'psmove' has no attribute 'PSMoveTracker'
```
Looking inside the `psmove.py` I see that only the `PSMove` class is defined, there's no definition for `PSMoveTracker`

nitsch, 2021-01-14:
No, this is not intentional. The tracker parts are missing from the Linux and Windows releases. They are present in the macOS release. Seems to be some kind of problem with the automatic CI builds.

nitsch, 2021-01-16:
@thp Can you look into this? At first glance it seems that the tracker parts in psmove.i guarded with `#ifdef PSMOVE_BUILD_TRACKER` are missing, i.e. passing `PSMOVE_BUILD_TRACKER` into the SWIG build does not work properly. I cannot reproduce this issue locally, but it is happening on the CI server.

keops-dev, 2022-07-18:
In build_debian I have to add cmake option `-DPSMOVE_BUILD_TRACKER=ON` to build with tracker

commit 0da66e0d (2022-10-14): Linux/Debian: Force building tracker (Fixes #437)
commit d499ac88 (2022-10-14): Merge pull request #451 from thp/enable-tracker-linux

Linux/Debian: Force building tracker (Fixes #437)
closed 2022-10-14 by thp