thp/psmoveapi #438: Error running MacOS Build

ChaoticPumpkin, 2021-01-19:
I've tried to open `psmoveapi-4.0.12-macos/bin/psmove` from the latest MacOS build of PS Move API, but when I run it, the terminal shows: 
```
dyld: Library not loaded: /Users/travis/build/thp/psmoveapi/build/libpsmoveapi.dylib
  Referenced from: /Users/silas/Downloads/psmoveapi-4.0.12-macos/bin/psmove
  Reason: image not found
[1]    30793 abort      /Users/silas/Downloads/psmoveapi-4.0.12-macos/bin/psmove
```
I also tried to compile it myself, but I'm still kinda a noob when it comes to the terminal and compiling, and couldn't figure out what i was doing wrong. I've been trying to figure this out for a while, hopefully someone can help.

ChaoticPumpkin, 2021-01-20:
Alright I compiled it correctly, should still look into the error on the build unless I was missing something myself

triscuitcircuit, 2021-01-30:
Im getting the same error. What did you do to fix it?

ChaoticPumpkin, 2021-01-30:
> Im getting the same error. What did you do to fix it?

I wasn't able to figure how to fix it, after a while of trying different things I was able to compile it myself, but not sure how to fix the error on the build, sorry

triscuitcircuit, 2021-01-30:
Did you compile it form source? Im also having an issue compiling from source, but thats probably an issue from my setup. Thanks for replying tho

ChaoticPumpkin, 2021-01-30:
Yeah, I did compile from the source, I wish I could help you do it, but honestly I don't remember how I did it now /:

thp, 2021-02-01:
Try setting `DYLD_LIBRARY_PATH` to where your `libpsmoveapi.dylib` is, this might help.

In theory, the rpath should be set properly, but maybe that broke at some point.

triscuitcircuit, 2021-02-03:
Everytime I try building from source, I get a clang linker error. However, I might just need to switch the terminal mode to Rosetta 2, which is weird cause the libraries compile fine on M1 Arm (at least from the install script).

Im going to try the posted suggestion as well to fix the pre-compiled version looking at the wrong path. 
```[ 10%] Building C object CMakeFiles/psmoveapi.dir/src/psmove.c.o
Undefined symbols for architecture x86_64:
  "_SDL_Delay", referenced from:
      _main in test_navcon.cpp.o
  "_SDL_Init", referenced from:
      _main in test_navcon.cpp.o
  "_SDL_JoystickClose", referenced from:
      _main in test_navcon.cpp.o
  "_SDL_JoystickGetAttached", referenced from:
      _main in test_navcon.cpp.o
  "_SDL_JoystickGetAxis", referenced from:
      _main in test_navcon.cpp.o
  "_SDL_JoystickGetButton", referenced from:
      _main in test_navcon.cpp.o
  "_SDL_JoystickNameForIndex", referenced from:
      _main in test_navcon.cpp.o
  "_SDL_JoystickOpen", referenced from:
      _main in test_navcon.cpp.o
  "_SDL_JoystickUpdate", referenced from:
      _main in test_navcon.cpp.o
  "_SDL_NumJoysticks", referenced from:
      _main in test_navcon.cpp.o
  "_SDL_Quit", referenced from:
      _main in test_navcon.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [test_navcon] Error 1
make[1]: *** [CMakeFiles/test_navcon.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....```

thp, 2021-02-06:
```
ld: symbol(s) not found for architecture x86_64
```

Could it be that your SDL2 library (probably from Homebrew) was compiled for aarch64 only? If it's just the navcon test, and you don't care about it, you can comment it out in CMakeLists.txt.

I don't think anyone from the PS Move API team has a M1 Mac yet for testing this, so you're unfortunately on your own here at the moment. With that said, it would probably work if you tried to build all dependencies + PS Move API for aarch64, I don't see any reason why it wouldn't build on 64-bit ARM (it definitely works on 64-bit x86 and 32-bit ARM, so there shouldn't be any pointer size or CPU architecture issues, and it's all little-endian).

FreakyBigFoot, 2021-12-07:
@ChaoticPumpkin So were you able to get this working on a M1 mac? I've been trying at this all evening and I'm stuck.

thp, 2022-08-29:
Marking as duplicate of #440, see there.

closed 2022-08-29 by thp