thp/psmoveapi #439: Many of the Python Examples have missing parentheses

SilasStewart, 2021-01-20:
In many of the python examples, running them will give errors saying: 
```
  File "/psmoveapi/examples/python/tracker_image_pygame.py", line 54
    print 'Trying to calibrate...'
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print('Trying to calibrate...')?
```
This is a pretty simple fix to add them, but I can see how working with multiple languages can get hard to get the syntax correct

nitsch, 2021-01-20:
> This is a pretty simple fix to add them, but I can see how working
> with multiple languages can get hard to get the syntax correct

What are you talking about? This is old Python 2 syntax. That code
simply is rather old.

SilasStewart, 2021-01-23:
Oh, whoops, I completely missed that! Sorry to bother you.

closed 2021-01-23 by SilasStewart