thp/psmoveapi #440: Building from source on Mac M1 issue

triscuitcircuit, 2021-02-07:
With information provided by issue #438 , I set `option(PSMOVE_BUILD_NAVCON_TEST "Build the navcon C test" OFF)` in CMakeLists.txt file within the examples folder. With this set to `ON` it would have a `ld` error at 40% completion (as in the orignal post) , with it `OFF` it has an ld error at 70% completion. 
```
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libpsmoveapi_tracker.dylib] Error 1
make[1]: *** [CMakeFiles/psmoveapi_tracker.dir/all] Error 2
make: *** [all] Error 2 
```
(Note: This happens regardless of the architecture set in the `build-macos.sh` script if that is of any help)
<details>
<summary>Full make output</summary>
<pre> [ 70%] Linking CXX shared library libpsmoveapi_tracker.dylib
ld: warning: ignoring file /opt/homebrew/lib/libusb-1.0.a, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
Undefined symbols for architecture x86_64:
  "_libusb_alloc_transfer", referenced from:
      ps3eye::URBDesc::start_transfers(libusb_device_handle*, unsigned int) in ps3eye.cpp.o
  "_libusb_cancel_transfer", referenced from:
      ps3eye::URBDesc::close_transfers() in ps3eye.cpp.o
  "_libusb_claim_interface", referenced from:
      ps3eye::PS3EYECam::init(unsigned int, unsigned int, unsigned char, ps3eye::PS3EYECam::EOutputFormat) in ps3eye.cpp.o
      ps3eye::PS3EYECam::open_usb() in ps3eye.cpp.o
  "_libusb_clear_halt", referenced from:
      ps3eye::URBDesc::start_transfers(libusb_device_handle*, unsigned int) in ps3eye.cpp.o
  "_libusb_close", referenced from:
      ps3eye::USBMgr::listDevices(std::__1::vector<std::__1::shared_ptr<ps3eye::PS3EYECam>, std::__1::allocator<std::__1::shared_ptr<ps3eye::PS3EYECam> > >&) in ps3eye.cpp.o
      ps3eye::PS3EYECam::~PS3EYECam() in ps3eye.cpp.o
      ps3eye::PS3EYECam::release() in ps3eye.cpp.o
      ps3eye::PS3EYECam::close_usb() in ps3eye.cpp.o
  "_libusb_control_transfer", referenced from:
      ps3eye::PS3EYECam::~PS3EYECam() in ps3eye.cpp.o
      ps3eye::PS3EYECam::stop() in ps3eye.cpp.o
      ps3eye::PS3EYECam::init(unsigned int, unsigned int, unsigned char, ps3eye::PS3EYECam::EOutputFormat) in ps3eye.cpp.o
      ps3eye::PS3EYECam::ov534_set_frame_rate(unsigned char, bool) in ps3eye.cpp.o
      ps3eye::PS3EYECam::ov534_reg_write(unsigned short, unsigned char) in ps3eye.cpp.o
      ps3eye::PS3EYECam::sccb_reg_write(unsigned char, unsigned char) in ps3eye.cpp.o
      ps3eye::PS3EYECam::sccb_reg_read(unsigned short) in ps3eye.cpp.o
      ...
  "_libusb_exit", referenced from:
      ps3eye::USBMgr::~USBMgr() in ps3eye.cpp.o
      ps3eye::USBMgr::~USBMgr() in ps3eye.cpp.o
      ps3eye::USBMgr::instance() in ps3eye.cpp.o
      std::__1::__shared_ptr_pointer<ps3eye::USBMgr*, std::__1::default_delete<ps3eye::USBMgr>, std::__1::allocator<ps3eye::USBMgr> >::__on_zero_shared() in ps3eye.cpp.o
  "_libusb_free_config_descriptor", referenced from:
      ps3eye::URBDesc::start_transfers(libusb_device_handle*, unsigned int) in ps3eye.cpp.o
  "_libusb_free_device_list", referenced from:
      ps3eye::USBMgr::listDevices(std::__1::vector<std::__1::shared_ptr<ps3eye::PS3EYECam>, std::__1::allocator<std::__1::shared_ptr<ps3eye::PS3EYECam> > >&) in ps3eye.cpp.o
  "_libusb_free_transfer", referenced from:
      ps3eye::transfer_completed_callback(libusb_transfer*) in ps3eye.cpp.o
  "_libusb_get_active_config_descriptor", referenced from:
      ps3eye::URBDesc::start_transfers(libusb_device_handle*, unsigned int) in ps3eye.cpp.o
  "_libusb_get_device", referenced from:
      ps3eye::URBDesc::start_transfers(libusb_device_handle*, unsigned int) in ps3eye.cpp.o
  "_libusb_get_device_descriptor", referenced from:
      ps3eye::USBMgr::listDevices(std::__1::vector<std::__1::shared_ptr<ps3eye::PS3EYECam>, std::__1::allocator<std::__1::shared_ptr<ps3eye::PS3EYECam> > >&) in ps3eye.cpp.o
  "_libusb_get_device_list", referenced from:
      ps3eye::USBMgr::listDevices(std::__1::vector<std::__1::shared_ptr<ps3eye::PS3EYECam>, std::__1::allocator<std::__1::shared_ptr<ps3eye::PS3EYECam> > >&) in ps3eye.cpp.o
  "_libusb_handle_events_timeout_completed", referenced from:
      ps3eye::USBMgr::transferThreadFunc() in ps3eye.cpp.o
  "_libusb_init", referenced from:
      ps3eye::USBMgr::USBMgr() in ps3eye.cpp.o
      ps3eye::USBMgr::USBMgr() in ps3eye.cpp.o
      ps3eye::USBMgr::instance() in ps3eye.cpp.o
  "_libusb_open", referenced from:
      ps3eye::USBMgr::listDevices(std::__1::vector<std::__1::shared_ptr<ps3eye::PS3EYECam>, std::__1::allocator<std::__1::shared_ptr<ps3eye::PS3EYECam> > >&) in ps3eye.cpp.o
      ps3eye::PS3EYECam::init(unsigned int, unsigned int, unsigned char, ps3eye::PS3EYECam::EOutputFormat) in ps3eye.cpp.o
      ps3eye::PS3EYECam::open_usb() in ps3eye.cpp.o
  "_libusb_ref_device", referenced from:
      ps3eye::USBMgr::listDevices(std::__1::vector<std::__1::shared_ptr<ps3eye::PS3EYECam>, std::__1::allocator<std::__1::shared_ptr<ps3eye::PS3EYECam> > >&) in ps3eye.cpp.o
  "_libusb_release_interface", referenced from:
      ps3eye::PS3EYECam::~PS3EYECam() in ps3eye.cpp.o
      ps3eye::PS3EYECam::release() in ps3eye.cpp.o
      ps3eye::PS3EYECam::close_usb() in ps3eye.cpp.o
  "_libusb_set_debug", referenced from:
      ps3eye::USBMgr::USBMgr() in ps3eye.cpp.o
      ps3eye::USBMgr::USBMgr() in ps3eye.cpp.o
      ps3eye::USBMgr::instance() in ps3eye.cpp.o
  "_libusb_submit_transfer", referenced from:
      ps3eye::URBDesc::start_transfers(libusb_device_handle*, unsigned int) in ps3eye.cpp.o
      ps3eye::transfer_completed_callback(libusb_transfer*) in ps3eye.cpp.o
  "_libusb_unref_device", referenced from:
      ps3eye::PS3EYECam::~PS3EYECam() in ps3eye.cpp.o
      ps3eye::PS3EYECam::release() in ps3eye.cpp.o
      ps3eye::PS3EYECam::close_usb() in ps3eye.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libpsmoveapi_tracker.dylib] Error 1
make[1]: *** [CMakeFiles/psmoveapi_tracker.dir/all] Error 2
make: *** [all] Error 2 </pre>
</details>

thp, 2021-02-12:
You need to build libusb as universal2 library, it says that right on the top:
```
ld: warning: ignoring file /opt/homebrew/lib/libusb-1.0.a, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
```

triscuitcircuit, 2021-07-17:
Hello thanks for the response! Sorry for the late reply, life has been hectic few months. 

I changed everything in the build script to be "arm" instead of "x86-64" and downloaded the most recent libusb with brew and compiled it for arm. 

However i still get this error with `make`:
```
23 warnings generated.
[ 43%] Linking C static library libglfw3.a
cd /Volumes/MacStorage/github/psmoveapi/build/external/glfw/src && /opt/homebrew/Cellar/cmake/3.21.0/bin/cmake -P CMakeFiles/glfw.dir/cmake_clean_target.cmake
cd /Volumes/MacStorage/github/psmoveapi/build/external/glfw/src && /opt/homebrew/Cellar/cmake/3.21.0/bin/cmake -E cmake_link_script CMakeFiles/glfw.dir/link.txt --verbose=1
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ar qc libglfw3.a CMakeFiles/glfw.dir/context.c.o CMakeFiles/glfw.dir/init.c.o CMakeFiles/glfw.dir/input.c.o CMakeFiles/glfw.dir/monitor.c.o CMakeFiles/glfw.dir/vulkan.c.o CMakeFiles/glfw.dir/window.c.o CMakeFiles/glfw.dir/cocoa_init.m.o CMakeFiles/glfw.dir/cocoa_joystick.m.o CMakeFiles/glfw.dir/cocoa_monitor.m.o CMakeFiles/glfw.dir/cocoa_window.m.o CMakeFiles/glfw.dir/cocoa_time.c.o CMakeFiles/glfw.dir/posix_tls.c.o CMakeFiles/glfw.dir/nsgl_context.m.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib libglfw3.a
[ 43%] Built target glfw
make: *** [all] Error 2

```

Thanks! I was looking at using this project for my custom game engine!!!

thp, 2021-07-18:
Please post the full build log.

triscuitcircuit, 2021-07-19:
It was the same issue of it not linking with libusb properly. Sorry I didnt state that earlier... Heres what I tried:

I installed the most recent libusb from brew and disabled C tests via the cmake before the build.

I also modified the Mac build script so it should, *theoretically* , builds the universal binary version ( as suggested) by using both flags: export `ARCHFLAGS='-arch x86_64 -arch arm64'` and `-DCMAKE_OSX_ARCHITECTURES="arm64;x86_64"`  when needed in the script. 

However I still get stuck with this error:

`ld: warning: ignoring file /opt/homebrew/lib/libusb-1.0.a, building for macOS-x86_64 but attempting to link with file built for macOS-arm64`

thp, 2021-07-29:
You have to build libusb from Homebrew as universal binary (shared library most likely). As the message says, `/opt/homebrew/lib/libusb-1.0.a` is a static library that was built for 64-bit ARM only.

triscuitcircuit, 2021-07-30:
Thank you for the reply!

From what i could find there's no current way to build a universal2 static library from Homebrew. The `--universal` flag is meant for the PowerPC/Intel Mac transition, and I cant find any documentation on a new version. 

The only thing i could find on this topic was this github [issue](https://github.com/Homebrew/brew/issues/10307)  where it was stated "By default will double the built time and require changes in almost every formula for this to work."

I cant find any documentation on how to build libusb from Homebrew as a Universal2 binary

thp, 2021-07-31:
Just build libusb from source then, not via Homebrew?

thp, 2021-07-31:
In any case, if you don't get it working, I'll fix it eventually when I get my hands on an M1 Mac, it's just difficult to debug this with just snippets of log files. The code definitely works on 64-bit ARM (tested on 64-bit ARM Linux).

thp, 2022-08-29:
I pushed a change to the `master` branch that should auto-detect macOS running on Apple Silicon (M1).

Clone the repo, make sure you have ARM64 Homebrew dependencies (e.g. libusb, libusb-compat, ...) installed, then run:

```
bash -e -x scripts/macos/build-macos
```

This should leave you with ARM64 binaries:

```
% file build/psmove build/libpsmoveapi*
build/psmove:                     Mach-O 64-bit executable arm64
build/libpsmoveapi.dylib:         Mach-O 64-bit dynamically linked shared library arm64
build/libpsmoveapi_tracker.dylib: Mach-O 64-bit dynamically linked shared library arm64
```

commit c6847fd2 (2022-08-29): macOS: Add build support for M1/Apple Silicon (Fixes #440)
closed 2022-08-29 by thp