thp/psmoveapi #441: build-debian fails with missing /external/hidapi/linux/hid.c

robtrooper, 2021-03-06:
System
Linux rob-desktop 5.4.0-66-generic #74-Ubuntu SMP Wed Jan 27 22:54:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

PS Move API 4.0.12

Cloned repo and ran 
bash -e -x scripts/install_dependencies.sh
without issues.

Compiling with
bash -e -x scripts/linux/build-debian
failed. 
last dozen lines of output-

`  Language bindings
    Python:           Yes (version 3.8)
    Java:             Yes
    C#:               Yes
    Processing:       Yes

-- Configuring done
CMake Error at src/CMakeLists.txt:166 (add_library):
  Cannot find source file:

    /home/rob/psmoveapi/external/hidapi/linux/hid.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  CMakeLists.txt:17 (include)

CMake Error at src/CMakeLists.txt:166 (add_library):
  No SOURCES given to target: psmoveapi
Call Stack (most recent call first):
  CMakeLists.txt:17 (include)

CMake Generate step failed.  Build files cannot be regenerated correctly.`

Looking in the path
/home/rob/psmoveapi/external/hidapi/linux/hid.c

The directory /hidapi/ is empty.

I can see that the github folder links to 
https://github.com/thp/hidapi/tree/38681c734965636c9fd710dc3898d90aaf923bb9

but is not handling this.

thp, 2021-03-06:
In the root checkout:
```
git submodule update --init --recursive
```

robtrooper, 2021-03-06:
That did it. Thanks.

closed 2021-03-06 by thp