thp/psmoveapi #442: Possible Yaw Drift

a-eye-tech, 2021-04-04:
I think I might be experiencing yaw drift. Here's a video I made: https://youtu.be/fZwU3tOpcNo

Whatever the problem is, how can I fix it? Does the psmoveapi have a function that can help me fix this issue?

yuripourre, 2021-04-04:
@a-eye-tech does your controller is calibrated? Does your notebook has speakers?

Have you tried another surface?

a-eye-tech, 2021-04-04:
@yuripourre I'm not sure if it's calibrated by the psmoveapi. I used my own code to have it oriented properly with respect to the monitor, which works for the first few seconds before it starts to drift. 

My notebook does have speakers.

I tried it on my desktop computer as well, and it has the same drift problem.

yuripourre, 2021-04-04:
@a-eye-tech one simple test is to point your controller into all directions 360°. If you search for "How to calibrate a gyro" in Google you might find something useful.

If the problem persists is can be related to the code.

a-eye-tech, 2021-04-13:
Ok, I fixed my problem. Here's what I did:

* Found the location of psmove.exe.
* Open Command Prompt in that directory, which for me was <...>psmoveapi\out\build\x64-Debug (default)
* On Command Prompt, run 'psmove.exe calibrate'
* Follow the instructions to have the magnetometer calibrated.

yuripourre, 2021-04-13:
@a-eye-tech glad that you solved it!

closed 2021-04-15 by a-eye-tech