thp/psmoveapi #444: Can't build on Mac (logs attached)

UjjwalGupta-Git, 2021-04-15:
The build-macos script didn't run successfully and don't know much about this so I uploaded the logs as the console said. Please Help.

[CMakeError.log](https://github.com/thp/psmoveapi/files/6316417/CMakeError.log)
[CMakeOutput.log](https://github.com/thp/psmoveapi/files/6316420/CMakeOutput.log)

UjjwalGupta-Git, 2021-04-15:
By the way I am on macOS Big Sur 11.2.1

thp, 2021-04-15:
This error:

```
Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
```

Did you try installing it with:

```
brew install pkg-config
```

This should normally be installed (as indirect dependency) with this script: `scripts/install_dependencies.sh`

Are you running on an Intel or Apple Silicon Mac?

UjjwalGupta-Git, 2021-04-16:
The documentation didn't say anything about this pkg-config and I am using Intel mid 2015 Macbook Pro

UjjwalGupta-Git, 2021-04-17:
So I deleted my build folder and then ran `brew install pkg-config`
then I got an error with the build-macos script about GNU m4. Then I looked up and found out that I had to update my developer tools. So I ran `xcode-select install` and ran the build-macos script again and everything worked perfectly. Thank you so much for your help.

closed 2021-04-17 by UjjwalGupta-Git